micropython/micropython #2615: extmod/machine_i2c uses optional HW API feature (OPEN_DRAIN), doesn't work for all ports/boards

pfalcon, 2016-11-09:
We now have 2 ports (Zephyr, Linux) which don't implement Pin.OPEN_DRAIN mode, which is well known to be optional (simply because not all hardware supports it per its specification; actually, not all software does that either). So, it's proposed to rewrite the driver to utilize only standard HW API features (Pin.IN, Pin.OUT).

pfalcon, 2016-11-09:
This ticket continues sustainable HW API rectification process started a year and a thousand tickets ago in
https://github.com/micropython/micropython/pull/1611 . Then, esp8266 port was supposed to be model port to show HW API "levels" and expected development process (mandatory features enable major functionality, optional features are merely optional), but hasted implementation of OPEN_DRAIN for it broke all the fun.

Well, we're now back to square one: people don't implement OPEN_DRAIN, as can be seen by 2 ports above (disclosure: Zephyr Pin class implementation wasn't done by me), nor its needed to implement bitbanged I2C.

dpgeorge, 2016-11-10:
This was discussed previously and a resolution made to use simple macros to emulate open-drain function calls, see https://github.com/micropython/micropython/issues/1430#issuecomment-221836971 and https://github.com/micropython/micropython/issues/1430#issuecomment-221899212

pfalcon, 2016-11-10:
Ok, I'll get to testing/implementing that resolution.

chrismas9, 2016-11-10:
Please don't reduce the Hardware API down to the lowest common hardware level or every potential MCU or port. It will  nobble MicroPython and prevent people using the capabilities of the newer or better MCU's. Hardware Open Drain is important for some applications and its loss would severely affect code and driver portability. Using the same argument for removing Open Drain (its not universally available) you would also have to delete I2C as its not in the ESP8266 hardware. When a function is needed, but not in all MCU's people will pick an MCU and port that supports that function, just like they would pick pyboard, not pyblite if they need CANbus or DAC.

I use open drain outputs in all sorts of ways; one of the most common is as a level translator and this is where portability comes in. Normally you would use pin.value(0 | 1) to drive a 3.3V compatible thing, say a red LED. To drive a blue LED from 5V you would have to use pin.mode(pin.IN | pin.OUT). A device driver for a character mode LCD would probably bitbang 4 bit of data, CS and WR using pin.value. But newer LCDs have a 3V limit. I run these from 2.9V (schottky diode drop from 3.3V - cheap regulator) with open drain pins and pullups to 2.9V. Its much easier to just instantiate the pins as open drain than rewrite the driver to emulate open drain with pin.IN and pin.OUT.

Another common use is multidrop wire OR'd links between boards for comms or status (eg low until the last board is not busy, then high for ready). I have an issue with the implementation of pin, especially for open drain. If you prefer I can discuss this elsewhere. Why is pin.value() undefined when the pin is driven (pin.OUT or pin.OPEN_DRAIN)? It is useful for a process to be able to read a 0 if the pin is low, irrespective if the low was asserted by the same board (undefined value) or by another board (will read 0). I know you can work around this, but it complicates the code and needs memory. I can't see any hardware reason why pin.value shouldn't be able to read a 0 or 1 from a driven pin or open drain pin.

I know you have resolved to keep Open Drain but I am concerned about the principle of abandoning or not supporting features because they are not universal.

dpgeorge, 2016-11-11:
> Hardware Open Drain is important for some applications and its loss would severely affect code and driver portability.

Don't worry, this issue is not about removing OPEN_DRAIN from the Python Pin API, it's about the internal C implementation of the SW I2C driver, using the C-level open-drain calls.

> Why is pin.value() undefined when the pin is driven (pin.OUT or pin.OPEN_DRAIN)?

It is defined if the pin is in OPEN_DRAIN mode and driven "high" (ie floating).  I think that covers your use-case.  Otherwise, please discuss this elsewhere and we can possibly refine the specs.

dpgeorge, 2018-12-04:
As discussed above, the issue raised in this ticket has a solution/resolution.

closed 2018-12-04 by dpgeorge