micropython/micropython #2616: round function faulty on esp8266

s4mdf0o1, 2016-11-10:
hi all,
here is a very simple issue :
on a built micropython for esp8266 :
in the REPL via USB UART :
```
>>> a=22.8654321
>>> round(a,1)
22.89999
```

while, on the same built micropython on my machine :
```
$ micropython # built micropython/unix/micropython
MicroPython v1.8.5-124-gbc4441a on 2016-11-07; linux version
>>> a=22.8654321
>>> round(a,1)
22.9
```

dpgeorge, 2016-11-10:
It's not the round function but rather a combination of the way 30-bit fp numbers are stored, and the way they are printed with the default precision settings.  See the same issue reported for the microbit: https://github.com/bbcmicrobit/micropython/issues/367

If you want to really print 1 decimal place then use string formatting: `'%.1f' % round(a, 1)`.  Otherwise this is not an easy thing to fix in the code.

s4mdf0o1, 2016-11-10:
many thanks for the alternative method :) ;)

peterhinch, 2016-11-10:
@s4mdf0o1 It's bound to do that if you convert it back to a float. Floating point numbers have limited precision; this means that certain values can't be exactly represented. @dpgeorge was suggesting a way to convert it to a string for display.

dpgeorge, 2017-04-18:
A very simple way to fix this is to just change the default precision for printing float/complex numbers (ie that done by str or repr of a float object).  Currently it's set at 7 for builds with float, and 16 for double.  Changing it to 6 for builds that use 30-bit FP representation fixes all the problems here and in the other referenced issues.

PR #3029 (2017-04-21): py: Reduce str/repr precision of float numbers when floats are 30-bit.
With 30-bit floats there aren't enough bits to faithfully print 7 decimal digits, so reduce the precision to 6 digits.

Fixes #2616.
closed 2017-04-21 by dpgeorge