micropython/micropython #2617: Improve docs

iBobik, 2016-11-10:
Sometimes I can not find some library in docs but I see code in this repository.

For example:
- DHT: [docs](http://docs.micropython.org/en/latest/pyboard/search.html?q=dht&check_keywords=yes&area=default#), [code](https://github.com/micropython/micropython/blob/master/esp8266/scripts/dht.py)
- ntptime: [docs](http://docs.micropython.org/en/latest/pyboard/search.html?q=ntptime&check_keywords=yes&area=default#), [code](https://github.com/micropython/micropython/blob/master/esp8266/scripts/ntptime.py)
- and probably more.

It seems something is wrong in docs generating.

mcauser, 2016-11-11:
First issue, you're searching for "DHT" in the pyboard docs, but it is currently only in the ESP8266 port. Use the menu in the bottom left to switch to ESP8266 and [search again](http://docs.micropython.org/en/latest/esp8266/search.html?q=dht&check_keywords=yes&area=default).

Second issue, I'm only seeing [quick ref](https://github.com/micropython/micropython/blob/master/docs/esp8266/quickref.rst) come up in the results for "DHT". If you search for "temperature", however, you see the [DHT tutorial page](https://github.com/micropython/micropython/blob/master/docs/esp8266/tutorial/dht.rst).

It seems there is an issue with search not finding the "DHT" on the [DHT tutorial page](http://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/dht.html?highlight=temperature), but it is finding "temperature" on the page.

Looks like ntptime hasn't been added to the docs yet.

iBobik, 2016-11-11:
Aha, little schizophrenia about switching boards. I thought it will search
in all ports.

Second issue: So docs are not auto-generated from the code? It should be,
shouldn’t be?

Jan Pobořil

mcauser, 2016-11-11:
Docs are generated from the code. I am not sure how frequently http://docs.micropython.org/ is updated though.

Instructions for building locally [here](https://github.com/micropython/micropython/blob/master/docs/README.md).

deshipu, 2016-11-11:
They are not generated from the code in the way you would generate docs from docstrings in Python code, because the C code doesn't have such straightforward correspondence between the things exposed on the Python side, and the code on the C side, and there is also no ready module for Sphinx for extracting comments from C code, as far as I'm aware.

pfalcon, 2017-06-30:
The scope of this ticket is too wide - there's always something to improve in the docs. Interested parties are welcome to submit (and actually do) specific suggestions, like https://github.com/micropython/micropython/issues/3170, https://github.com/micropython/micropython/issues/3172 , preferrably with the desire to back them with patches if approved.

closed 2017-06-30 by pfalcon