micropython/micropython #2619: docs: Restructure TOC and review article links.

hosaka, 2016-11-11:
It would be a good opportunity with the current docs work, to review the structure of table of contents. It has been noted that the nav bar contains duplicates (https://github.com/micropython/micropython/issues/2578#issuecomment-259054477), hard to read/navigate, with some pages being difficult to find:

[image]

I propose having the top level looking something along the lines of:

- Quick Reference
- General Information
- Tutorials
- Libraries
- Hardware
- Datasheets
- License
- Contents

Menu items themselves do not need to add pyboard/esp8266/etc into the menus or article titles. The "Ports and Versions" menu is a good indicator of what's currently being viewed.

pfalcon, 2016-11-11:
https://github.com/micropython/micropython/issues/2578 contains more comments on this issue.

pfalcon, 2017-08-23:
This should be fixed by now. If there're more issues/idea, please open separate ticket(s). (Like e.g. https://github.com/micropython/micropython/issues/3188)

closed 2017-08-23 by pfalcon