micropython/micropython #2620: RFC: offical API to save WiFi-connection parameter

puuu, 2016-11-11:
After @MarkR42 explained that the automatic storing of the WiFi-connection parameter can wear-out the flash of the ESP8266 many discussions came up: #2510, #2547, #2564 and http://forum.micropython.org/viewtopic.php?f=16&t=2631

This explains the importance of this issue.

In principle, the feature of saving the connection parameter is due to the ESP8266 SDK and strictly speaking not compatible to the existing HW API (https://github.com/micropython/micropython/wiki/Hardware-API#the-wlan-class). But on the other hand it provides a nice user experience. Also @pfalcon explained that this is a common feature in smart phones, computers, … So we have to expect, that the ESP8266 will not be the only device that supports saving of WiFi-connection parameters.
But, this side effects can lead to permanent hardware damage. Thus, we need a way to at least deactivate this feature.

Therefore, we have to find an agreement about the method, default behaviour and API names. That is the reason for this issue.

There are already three different possible methods described:

a) #2510 additional parameter `save_to_flash` in `wlan.connect()`
b) https://github.com/micropython/micropython/pull/2510#issuecomment-255714114 additional function `esp.persist_wlan(val)` which allows the user to turn on/off the feature of saving WiFi settings to flash
c) #2564 additional function `esp.save_config()` to save configuration to flash on demand

As default behaviour we have the possibilities:
- `wlan.connect()` and `wlan.active()` always save to flash. That will probably lead to permanent hardware damage and is not possible for devices that do not support the saving (e.g. the WiPy board). 
- `wlan.connect()` and `wlan.active()` don't always save to flash. This will break the user experience of the ESP8266 users.

What are your comments?

kfricke, 2016-11-11:
Nice summary, but I would like to add again: not letting `wlan.connect()` and `wlan.active()` not persist settings to flash does not break things. The need to persist manually (e.g. `esp.save_config()` )it what is different. After this the experience is equal.

kfricke, 2016-11-11:
Of course this is a RFC for an official API. Just wanted to add that the ESP SDK does also include the methods `wifi_station_set_auto_connect(uint8 <0 or 1>)` and `wifi_station_set_reconnect_policy(bool)`.

Could be a suggestion of what else to think about in an official API.

peterhinch, 2016-11-12:
I favour option b) with the default being to save to flash. This avoids breaking code and conforms to expected behaviour.
In applications which permanently connect, flash wear is insignificant because reconnection only occurs when WiFi communication is disrupted. This might be (say) twice  day, giving > ten years operation from a $2 device.
If the user opts to override the default the assumption is that they have read the docs and appreciate the consequences.

torwag, 2016-12-03:
Hi,

I would like to add that saving the parameters is NOT a standard feature. Smartphones, Laptops, etc. all handle this within software (the OS). Saving them to flash automagically, would mean you could stick an USB-wireless dongle in any PC and a connection would be established. 

TLTR: IMHO the network function is wrongly placed since it suggest that it behaves the same among all ports. 

The real problem here and that is were people start to disagree, we have the clash of two worlds or APIs.
Micropythons API (for now) does not include saving wireless credentials in flash (as it is not possible on many devices). On the ESP8266 this is standard. Thus, whatever we do, we break one user experience resp, one API. That is the reason we see now movement because there is no perfect solution.

In my opinion if we keep a generic network library,  we have to change the micropython api to save the wireless config automatically for all devices or we have to break with esp standard behaviour and do not save the config at all. 

Right now we have a non-standard micropython behaviour for esp in the network library, which should behave identically among all different devices (at least that is what I understood should network be based on its location).
That was one of the reason I suggested a more clear separation between SoCs and Board functions and a clear naming scheme which makes it clear to users what to expect and where to look for certain features. #2595 

In my opinion we need three layers, 
A generic layer which defines functions which are and will always be exact the same among all kind of SoCs, MCUs, Boards, Embedded, Unix, ports,  etc.
A SoC layer which defines all functions of a particular SoC or MCU family, which again is the same among all boards which uses this particular chip
A Board layer which defines particular implementations and addressing other functions on the board itself. Naturally, inheritance would allow to reuse much of the underlying layers to define higher level specialised functions. 
 
I already realised that this scheme is not perfect either. E.g. network-functions have to be placed within the soc (e.g. esp)  or board (e.g. pyboard) layer. That might confuse users as well, however, it would directly point them to the way the network function is implemented. Someone who calls an esp-soc function might not be surprised that this behaves different compared to a pyboard-board function.

For the particular case here. Now we have network.connect(args) for both esp and others however, with different behaviour which is bad in my opinion.
A function naming like esp.network.connect(args, persist_wlan=True) would clearly separate from a wipy.network.connect(args) which does not provide a persistent saving feature. Users would at least not expect identically behaviour since both functions come from a different class.

Hence, some sort of restructuring is needed especially, if micropython starts to spread to more and more boards and SoCs.

deshipu, 2017-08-08:
The ESP32 doesn't save its connection parameters. Should we make the ESP8266 compatible with all the other ports?

tekinozbek, 2019-10-16:
So... did we give up on this completely?

MrSurly, 2019-10-16:
> So... did we give up on this completely?

The thing is MP is a programming language.  The idea is the language and libraries provide you the tools to do it yourself

Adding features like this are just bloat.  I wouldn't use it even if it existed, I'd need something more robust.  I  use a `config.json` file.  It has stuff in there like certificates and keys for AWS access, cellular modem parameters, and WiFi credentials (yes plural).  The code uses this info to determine how to connect, including whether to prefer cell over wifi (or not), preferred wifi networks, and their order of connection attempts.

The core language should solve relatively low-level problems. Higher level problems should be accomplished in libraries or user end-code.

tekinozbek, 2019-10-16:
Huh? We are talking about preventing the hardware from writing your wifi credentials onto flash when you call connect(). It will wear down your flash if you try to deep-sleep then wake up every minute and call connect() like I do.

MicroPython is not a programming language, it is an implementation. And one that, apparently, does not give you access to the the low-level configurations of the hardware especially when the vendor already supports this in their C API.

Poor @kfricke screamed this problem for the better of this library but people said all sorts of irrelevant things to prevent it from being merged.

MrSurly, 2019-10-17:
> Huh? We are talking about preventing the hardware from writing your wifi credentials onto flash when you call connect().

Let's do that, and _also_ not add any option to turn it on (hence my bloat comment).

tekinozbek, 2019-10-17:
My bad -- I thought you were talking about implementation of credential storage as part of MicroPython (which, like you, I don't agree at all either).

However, I do see the concern with suddenly breaking userspace with such a change. In that case, it should be an opt-out feature and clearly written all across the documentation. Had I not been randomly googling ESP8266, I wouldn't have even known about this feature wrecking my flash when I'm deep sleeping every minute.

I was very happy at first to see that this issue was from 2016, thinking "surely, it's fixed by now" only to find that it's still open :(

peterhinch, 2019-10-17:
It is possible to code around this. You attempt to connect without specifying credentials. If that fails you then specify them. Thus you only write to flash when the credentials change.

From [the asynchronous MQTT repo](https://github.com/peterhinch/micropython-mqtt):
```python
    async def wifi_connect(self):
        s = self._sta_if
        if ESP8266:
            if s.isconnected():  # 1st attempt, already connected.
                return
            s.active(True)
            s.connect()  # ESP8266 remembers connection.
            while s.status() == network.STAT_CONNECTING:  # Break out on fail or success. Check once per sec.
                await asyncio.sleep(1)
            if not s.isconnected() and self._ssid is not None and self._wifi_pw is not None:
                s.connect(self._ssid, self._wifi_pw)
                while s.status() == network.STAT_CONNECTING:  # Break out on fail or success. Check once per sec.
                    await asyncio.sleep(1)
```

ricardomattar, 2021-08-11:
The lack of wifi.persistent is pityful. Its the only thing keeping me of using micropython.

EternityForest, 2022-04-21:
Seems like ESP32 does not do this, right? Wouldn't that be sign that Espressif sees persistence as a mistake?

Python is a high level language, that has JSON capabilities. It already gives you good ways to save credentials in a more flexible way yourself very easily.  I'm all for ditching it, things should not be able to sneakily do hardware damage in a very non-obvious way.

jonnor, 2024-07-19:
This can easily be solved in Python user code. There are several WiFi managers out there for MicroPython on Github. Proposing this is closed as a wont fix.