micropython/micropython #2621: Context manager __exit__() not called in generator which does not run to completion

peterhinch, 2016-11-12:
This occurs when a generator does not run to completion e.g. one yielding an infinite set. The following works in cPython but not in MicroPython:
```python
class foo(object):
    def __enter__(self):
        print('Enter')
    def __exit__(self, *args):
        print('Exit')

def bar(x):
    with foo():
        while True:
            x += 1
            yield x

def func():
    g = bar(0)
    for _ in range(3):
        print(next(g))

func()
```
MicroPython:
```
[adminpete@axolotl]: ~
$ unix/upython
MicroPython v1.8.1-794-g29f3f84-dirty on 2016-11-10; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import rats10
Enter
1
2
3
>>> 
```
cPython
```
$ python3
Python 3.4.3 (default, Sep 14 2016, 12:36:27) 
[GCC 4.8.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rats10
Enter
1
2
3
Exit
>>> 
```
A similar effect can be seen in try...finally blocks. In cPython the finally block runs on exit from the REPL, in MicroPython never. In my view the context manager issue is the more serious one.

pfalcon, 2016-11-12:
Not only that. Add `__del__` to `foo` and you'll see that it's not called (and that's the only thing which can call `__exit__` in your scenario). Evenmore if after that you try:

~~~
f = foo()
f = None
gc.collect()
~~~

You'll see that finalizer is not called at all, even though it should be on unix. We have no single test for `__del__`, so no wonder it's regressed. On the other hand, we're using conservative garbage collection, which means that there's not guarantee that all finalizers will be called at all.

if you want to have resources cleaned up, you'll need to do it yourself for now:

~~~
def func():
    g = bar(0)
    for _ in range(3):
        print(next(g))
    g.close()
~~~

The first in queue for alleviation of the problem aren't generators and context managers, but poor esp8266 sockets. Nobody closes them properly, I don't believe that. Well, I don't. Not when I play on command line, not when writing scripting. umqtt.robust doesn't close sockets, which means it's bound to crash ESP8266 SDK sooner or later. And nobody notices (the specific problem) - everybody got used that resources are free, infinite, and somebody else is taking care of them.

pfalcon, 2016-11-12:
> You'll see that finalizer is not called at all, even though it should be on unix. We have no single test for __del__, so no wonder it's regressed.

Heh, of course, we don't support `__del__` for Python objects, only for native objects. So, this feature will likely never be implemented. You have 2 approaches: 1) If "a generator does not run to completion", then well, it doesn't exit, so no wondering `__exit__` is not called, you quit the application, the generator still runs somewhere in a better world; 2) if you have infinite generator, but don't want it to run infinitely, well, stop it. It's the same as with sockets, etc.

peterhinch, 2016-11-12:
Thanks for the simple workround.

jonnor, 2024-09-29:
Issue still reproduces on MicroPython 1.24 Unix port.

dlech, 2024-09-29:
I would suggest to close this one unless we want to add some documentation on it. 

The fact the the example works in CPython is not by design, but rather a side effect of using reference counting for object lifetime.

There is a deferred PEP https://peps.python.org/pep-0533/  that explains the general issue here, but since the issue isn't fully solved in CPython yet, there isn't much to do in MicroPython other than recommend to implement your own handling to ensure that `.close()` is called on all generators when they go out of scope.

closed 2025-03-03 by jonnor