micropython/micropython #2624: compiler: Try to optimize "and"/"or" for constant values?

pfalcon, 2016-11-12:
Having code like:
~~~
if __debug__:
    print(1)
~~~

And compiling it with -O leads to unreachable block to be optimized out:

~~~
00 LOAD_CONST_NONE
01 RETURN_VALUE
~~~

Similar for:
~~~
DEBUG = const(0)

if DEBUG:
    print(1)
~~~

However, using logical connective 
~~~
DEBUG = const(0)

if DEBUG and __debug__:
    print(1)
~~~

leads to:
~~~
00 LOAD_CONST_SMALL_INT 0
01 STORE_NAME DEBUG
04 JUMP 15
07 LOAD_NAME print (cache=0)
11 LOAD_CONST_SMALL_INT 1
12 CALL_FUNCTION n=1 nkw=0
14 POP_TOP
15 LOAD_CONST_NONE
16 RETURN_VALUE
~~~

I.e. the constant logical expression is optimized to just jump, but unreachable block is not skipped.

pfalcon, 2016-11-12:
Also, if Python guarantees short-circuiting of logical connectives, `__debug__ and DEBUG` could be optimized regardless if DEBUG even a var (and would be safe for expressions).

pfalcon, 2016-11-12:
This comes from me trying to make logging available for uasyncio, but be disablable well.

pfalcon, 2016-11-13:
https://github.com/micropython/micropython-lib/commit/4ffaf83c0edaae54e59a335067da74d48d28e435

dpgeorge, 2016-11-14:
See #2630.

closed 2016-11-18 by dpgeorge