micropython/micropython #2625: Esp8266 garbage output after upgrading to 1.8.6

Cediddi, 2016-11-12:
Hi,
I have few esp modules I'm using with 1.8.5. With the latest release 1.8.6, I've been having this unusual state. Whenever I connect to esp via serial port, I get garbage output. The led on the board flashes rapidly. I tried all four latest official builds, stable, daily and 512k stable and 512k daily. All have the same output.

Uploading steps is this:
```
./esptool.py --port /dev/ttyUSB0 --baud 460800 erase_flash
./esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=8m 0 ~/Downloads/esp8266-20161110-v1.8.6.bin
sudo screen /dev/ttyUSB0 115200
```

My modules are 4MB

I'm also adding port_diag output.

```
FlashROM:
Flash ID: 1640e0 (Vendor: e0 Device: 4016)
Flash bootloader data:
Byte @2: 00
Byte @3: 40 (Flash size: 4MB Flash freq: 40MHZ)
Firmware checksum:
size: 563544
md5: 63a5d6aa1672329c2b9ad3dc69d23032
True

Networking:
STA ifconfig: ('0.0.0.0', '0.0.0.0', '0.0.0.0', '208.67.222.222')
AP ifconfig: ('192.168.4.1', '255.255.255.0', '192.168.4.1', '208.67.222.222')
Free WiFi driver buffers of type:
0: 8
1: 0
2: 8
3: 4
4: 7
lwIP PCBs:
Active PCB states:
Listen PCB states:
Local port 8266, foreign port 22442 snd_nxt 611362473 rcv_nxt 1073725424 State: LISTEN
TIME-WAIT PCB states:
```

Boards are Wemos D1mini, I'm not going to blame modules, as they're working perfectly under 1.8.5 and I already tried 6 of them.

Is there anyway else I can do to help you?

deshipu, 2016-11-12:
What version of esptool.py are you using?

robert-hh, 2016-11-13:
If your modules are 4MBytes sized, you should use --flash_size=32m with esptool.

Cediddi, 2016-11-13:
I pulled last esptool 1.3-dev and increased flash_size to 32m, it's working now, thanks :)

closed 2016-11-13 by Cediddi