micropython/micropython #2626: upip issues with esp8266?

mzdaniel, 2016-11-12:
First at all, thank you so much for this wonderful project! . Running python3 in a microcontroller like the esp8266 is just amazing.

I am using the open-source hardware Amica NodeMCU and flashed the latest 1.8.6 then, the daily micropython image on it and still getting errors.

This is what I am getting after a establishing network conectivity and later a hard reset:

```

could not open file 'main.py' for reading

MicroPython v1.8.6-9-g30cfdc2 on 2016-11-12; ESP module with ESP8266
Type "help()" for more information.
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 2 (4a0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with Seagull, channel 1                                    
dhcp client start...                                                 
ip:192.168.43.50,mask:255.255.255.0,gw:192.168.43.1                  
                                                                     
>>>                                                                  
>>> import upip
>>> upip.install('logging')
Installing to: /lib/
Warning: pypi.python.org SSL certificate is not validated
Installing logging 0.4.9.6 from https://pypi.python.org/packages/93/4b/979db9e44be09f71e85c9c8cfc42f258adfb7d93ce01deed2788b2948919/logging-0.4.9.6.tar.gz
Traceback (most recent call last):                                   
  File "<stdin>", line 1, in <module>                                
  File "upip.py", line 205, in install                               
  File "upip.py", line 195, in install                               
  File "upip.py", line 165, in install_pkg                           
  File "upip.py", line 93, in install_tar                            
  File "upip.py", line 63, in save_file                              
  File "upip.py", line 60, in save_file                              
  File "upip_utarfile.py", line 37, in readinto                      
OSError: [Errno 22] EINVAL                                           
>>> LmacRxBlk:1                            
LmacRxBlk:1                                                          
```

Shouldn't this work?

mzdaniel, 2016-11-13:
Of course, this is trying to connect to pypi!  The package micropython-logging is there and working. Still hard to believe this can be achieved with so little resources.  Awesome job!

closed 2016-11-13 by mzdaniel

tvestelind, 2017-05-08:
Sorry for re-opening this but I'm having the same problem. What caused this and how was it solved?

Cheers,
Tomas

girtsf, 2017-07-16:
To save the headache for somebody else searching for the same error: you want `upip.install('micropython-logging')`, not `upip.install('logging')`.