micropython/micropython #2627: cc3200: Convert to use open-source flashing tool

pfalcon, 2016-11-13:
One of the big problems with cc3200 port maintenance is proprietary flashing procedure. Fortunately good people of the world did for cc3200 something similar that esptool.py did for esp8266: https://github.com/ALLTERCO/cc3200tool . I tested it briefly with samples for cc3200 Zephyr port (native Zephyr samples I mean, uPy not involved) and it appears to roughly work. Next step is trying that tool with a real-world application like uPy.

dhylands, 2016-11-13:
I've been using cc3200tool quite a bit lately. At work the project I'm currently working on is a combination of cc3200 and nRF51822, but I've been programming in rust.

So far, I've only used cc3200tool for flashing /sys/mcuimg.bin, but that's been working reasonably well. Occasionally I need to reset the procesor, but otherwise its been working well.

pfalcon, 2016-11-13:
> I've been using cc3200tool quite a bit lately. At work the project I'm currently working on is a combination of cc3200 and nRF51822, but I've been programming in rust.

Well, good news!

> So far, I've only used cc3200tool for flashing /sys/mcuimg.bin

Same here.

dpgeorge, 2016-11-13:
+1 for cc3200tool, would make life a lot easier.

dmartauz, 2016-11-13:
If I understand correctly, using this tool makes sense for people with Linux on their dev PCs or are there any  other benefits in comparison to UniFlash?

dhylands, 2016-11-14:
I tried setting up uniflash under linux, and I did eventually get it to work, but its a 32-bit tool and requires installing a bunch of 32-bit dependencies, and it was just an all-around a pain. Using cc32tool under linux is hands down much simpler.

Using uniflash under Windows should be fine, assuming they have a command line version. It's not difficult for the makefiles to support multiple flashing methods. We already have that for stmhal where you can flash via dfu or flash via SWD.

pfalcon, 2016-12-26:
I was able to successfully flash cc3200-launchxl with cc3200tool. However, it took quite a few time to figure out that was a successful, not a dead, flash due to https://github.com/micropython/micropython/issues/2725.

pfalcon, 2016-12-29:
Makefile targets using cc3200tool roughly similar to esp8266 added in 0748143a4c648ad7bcdbbf125d039ceb7490f4e6. There's support for erasing flash, flashing upy bootloader/firmware and separately to flash a vendor wifi servicepack. I tested them all (and even in that order, confirming that servicepack can be flashed/upgraded after upy). All that is of course using cc3200-launchxl, as confirmed by danicampora in https://github.com/micropython/micropython/issues/2725 , wipy is not upgradable via serial (normally; one should be able to short-circuit SOP2 testpoint on the board to a suitable level during reset to reflash it nonetheless).

What's left: update port README. Quick suggestions for that are welcome, my plan is to word the cc3200tool to be the supported way, and uniflash to be an alternative, supported solely on TI side.

(I commit all this stuff directly, as it comes as by-product of me evaluating/testing cc3200tool for dayjob. I don't see myself coming back to it any time soon when I finish with it, and I want to finish ASAP. Post-review, corrections, etc. are more than welcome.)

pfalcon, 2016-12-29:
README updated in b315d76b6b76a81a11d1456fa43e195b711e406b. I'm closing this ticket, but feel free to provide suggestions on improvements. (@dpgeorge: feel free to patch it as you see fit.)

closed 2016-12-29 by pfalcon

dpgeorge, 2016-12-30:
Ok, thanks.  I won't have anything useful to say/contribute until I actually test cc3200tool, which might not be for some time.