micropython/micropython #2628: Extend obj->iternext to handle not just __next__(), but also send() method

pfalcon, 2016-11-13:
For generators (generalization of iterators), `.__next__()` method is defined as `.send(None)`. It's only natural to extend C-level `->iternext()` method to accept send argument, to allow implementation of C-level generators optimally. Caveat: so far, none of such exist. But work on optimizing uasyncio may lead to need for such. Or maybe not, maybe "virtual" awaitable methods will be used instead: https://github.com/micropython/micropython/issues/2622 .

This ticket is created to be referenced by in-code TODO comments (yes, I think we reached that code complexity that need to reference tickets straight from code).

pfalcon, 2016-11-13:
Other generator-related optimizations: 

https://github.com/micropython/micropython/issues/402
https://github.com/micropython/micropython/issues/369

pfalcon, 2016-11-13:
https://github.com/micropython/micropython/issues/369#issuecomment-38556195 says we could have 3 args for ->iternext at all.

pfalcon, 2016-11-14:
Here's commit which references this issue: 79d996a57b351e0ef354eb1e2f644b194433cc73

om2pkral, 2016-12-15:
Here is what worked for me: https://github.com/om2pkral/micropython/commit/6a27c72dbae05386d9a645d906061a06a4847130

It's halfway as there are some (and possible more than I think) issues.

- is_native_generator flag should be definitely implemented more generically
- type of a such generator is untouched

thinking about new mp_type_gen_native.