micropython/micropython #2629: machine.time_pulse_us should not throw exceptions

pfalcon, 2016-11-13:
I don't know how we ended up with throwing exceptions from such a function in the first place, but that needs to be changed to returning -1: throwing exceptions means allocation, and that function is too low-level and time-sensitive, can be used to decode modulated signals for example. It's ok if it can't decode first repetition in a pack of 5 (that's why repetition), it's not ok to throw an exception, trigger gc and miss all 5.

dpgeorge, 2016-11-21:
Following standard Python practices, "exceptional events" should raise an exception and a timeout is generally regarded as exceptional, ie not something you encounter in normal behaviour.

> that function is too low-level and time-sensitive

That's true, it is low-level and could benefit from special treatment re timeout.  But if it returns -1 on timeout then all code using the function must check the return value, which adds extra code/overhead to the general case (instead with exceptions just slowing down the exceptional case).

I'd be +0.1 on changing it.  If it is changed then probably it needs to distinguish between the 2 different timeouts, waiting for the first and second edge.  Eg a return value of -1 means first edge not found, -2 means second edge not found.

pfalcon, 2017-02-04:
Thanks. Review posted as https://github.com/micropython/micropython/pull/2836

pfalcon, 2017-05-14:
Fixed by PR above.

closed 2017-05-14 by pfalcon