micropython/micropython #2635: stmhal: WDT hangs after programming

dhylands, 2016-11-14:
I discovered an interesting quirk related to the IWDG.

If I program my pyboard using pydfu, then trying this in the REPL immediately afterwards hangs:
```
>>> wdt = machine.WDT(0, 5000)
```
By hangs, I mean it never returns the REPL. I added some debug prints and it seems to stuck in the IWDG->SR & 1 loop and the value of SR is 3.

If I hit the RESET button, then the `wdt = machine.WDT(0, 5000)` works exactly as expected.

dpgeorge, 2016-11-15:
Confirmed.  This looks similar to issues with the RTC not starting ofter DFU programming, and requiring an rtc.init() call to get it working.

I guess programming leaves some registers in an undefined, not-reset state.

chuckbook, 2016-11-19:
This seems to result from ERRATUM 2.2 in  DocID022183 Rev 7. With any ICP method make sure to explicitly reset after programming. However I don't know if pydfu has the reset option.

dhylands, 2016-11-19:
pydfu triggers the "leave" DFU mode, so that your downloaded code starts running right away. I don't see anything in the DFU protocol that would allow it to reset.

chuckbook, 2016-11-21:
I used to change the restart address in pydfu.py from 0x0800 0000 to 0x0800 fff0 (aka middle of nowhere). This seems to finally trigger a reset. I have only tested it on a F7 processor.

dhylands, 2016-11-21:
080f fff0 appears to be just before the end of the 1 Mb mark.

The pyboard has 1 Mb of flash, so this corresponds to a valid address. I tried some other invalid addresses and the set_address function fails.

So then I tried adding the following code to the startup_stm32.s file, immediately after the vector table:
```
  .align 9
  .word 0xffffffff
  .word 0xffffffff
```
and set the reset address to 0x8000200 and that also seems to work. I created PR https://github.com/micropython/micropython/pull/2646 which seems to work on the PYBV10.

hardkrash, 2020-11-10:
I ran into a similar issue, it appears that if the internal low speed oscillator is not enabled, the system will hang.

If the watchdog is enabled and the LSI is not enabled and ready than the code for enabling the watchdog will stick into the busy loops.

Bit 1 of register `RCC_CSR` is `LSIRDY` and bit 0 is `LSION`

Fixing this by setting LSION after a DFU flash doesn't fix this issue, but ensuring this is set prevents blocking in normal code.