micropython/micropython #2637: STM32L476 dfu upgrade bug

shaoziyang, 2016-11-15:
When I upgrade firmware for STM32L476 using dfu file ,it is not run after download.

But when I upgrade firmware using hex file, it works.

dhylands, 2016-11-15:
This appears to be an issue with dfu-util and not micropython. It also appears to be on their radar, but no specific fix has been posted yet: https://sourceforge.net/p/dfu-util/tickets/27/

I was able to reproduce the problem (i.e. if I flash my STM32L476 using dfu-util then it doesn't boot properly). However, if I flash the exact same image using pydfu.py then everything seems to be working fine.

dpgeorge, 2016-11-16:
Yes, this is an issue for dfu-util.  I would suggest using tools/pydfu.py from this repository, which is the default if you do "make deploy" in stmhal directory.  You can also use an st-link device.

closed 2016-11-16 by dpgeorge