micropython/micropython #2638: unix build error

adrianlzt, 2016-11-15:
```
% git clone https://github.com/micropython/micropython.git
Cloning into 'micropython'...
remote: Counting objects: 46654, done.
remote: Compressing objects: 100% (132/132), done.
remote: Total 46654 (delta 54), reused 0 (delta 0), pack-reused 46522
Receiving objects: 100% (46654/46654), 27.19 MiB | 3.20 MiB/s, done.
Resolving deltas: 100% (33507/33507), done.
% cd micropython/unix 
% git log -1
commit 30bca45e1ac9fc6953728950695135b491403215
Author: Damien George <damien.p.george@gmail.com>
Date:   Tue Nov 15 16:48:29 2016 +1100

    tests/basics: Add test for logical constant folding.
% make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: *** No rule to make target 'lib/berkeley-db-1.xx/btree/bt_close.c', needed by 'build/genhdr/qstr.i.last'.  Stop.

```

robert-hh, 2016-11-15:
After cloning mycroython, you must also add:
```
Building Unix version requires some dependencies installed. For
Debian/Ubuntu/Mint derivative Linux distros, install `build-essential`
(includes toolchain and make), `libffi-dev`, and `pkg-config` packages.

Other dependencies can be built together with MicroPython. Oftentimes,
you need to do this to enable extra features or capabilities. To build
these additional dependencies, first fetch git submodules for them:

    $ git submodule update --init

Use this same command to get the latest versions of dependencies, as
they are updated from time to time. After that, in `unix/` dir, execute:

    $ make deplibs

This will build all available dependencies (regardless whether they
are used or not). If you intend to build MicroPython with additional
options (like cross-compiling), the same set of options should be passed
to `make deplibs`. To actually enabled use of dependencies, edit
`unix/mpconfigport.mk` file, which has inline descriptions of the options.
For example, to build SSL module (required for `upip` tool described above),
set `MICROPY_PY_USSL` to 1.

```
(like told in README.md)

hosaka, 2016-11-15:
The Makefile rule that runs `submodule update --init --recursive` in https://github.com/micropython/micropython/blob/master/unix/Makefile#L278 could perhaps be changed to trigger automatically upon other make targets. But I think there was a comment somewhere against making deplibs automatically.

prusnak, 2016-11-15:
I run just `make axtls` after cloning with --recursive, because I prefer using system-wide libffi, not local one.

hosaka, 2016-11-16:
There's no need to be mean Paul. He got the answer and will close the issue in due time.

kfricke, 2016-11-16:
@adrianlzt please be not offended by special maintainer treatment. These are stressful times.

adrianlzt, 2016-11-18:
Wow, sorry for that.
I was following the make instructions for Arch https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=micropython but seems they are outdated.

Thank you all (except one :)

closed 2016-11-18 by adrianlzt

hosaka, 2016-11-18:
The best place for up to date instructions is the official docs, github and
our forum! Feel free to come along and join the discussion or of you need
help!