micropython/micropython #2642: esp8266: set default softAP False (OFF)

profra, 2016-11-17:
Here is probably better place to discuss my suggestion I mentioned on http://forum.micropython.org/viewtopic.php?f=16&t=2631&start=70#p15831 only as a note.

"Note: 
I would like to suggest (to @pfalcon) not to activate default softAP but to make it during WEBREPL activation. I think it's useful because I can see a room full of attendees during a training with fresh flashed MicroPython who shall solve the task BLINK_LED, while there is the air full of RF smog with e.g. activated 10 APs..., which are useless at that moment. When the trained people reach WiFi topics, they will be able to activate necessary functions."
... there are plenty of other reasons too.

kfricke, 2016-11-17:
Also see #2569

hosaka, 2016-11-22:
Although I agree with you and @kfricke, I don't use the AP all that much myself. We mustn't forget that the esp port is made as user friendly as possible: you power it on, you connect to it's AP, you run webrepl. If one doesn't own a usb-serial device, it would make it practically impossible to get to REPL. So, as you suggested to make soft AP during webrepl activation, how does one call webrepl in this situation?

Or have we disabled webrepl on by default now and I'm completely wrong?

profra, 2016-11-22:
Yes, you are right, webrepl has to be initialised through USB-serial device... but mostly is this on board, you need only suitable USB cable.... see http://forum.micropython.org/viewtopic.php?f=16&t=2140&sid=3dae2fe6afbf0004f4cbce5b8bb89c8a ... For this reason ... IMHO... , it makes no sense default SoftAP ON.

hosaka, 2016-11-22:
Okay, if we no longer have webrepl on by default. I think it makes sense to disable AP.

jonnor, 2024-07-19:
Is this still relevant?