micropython/micropython #2643: stmhal: I2C not working with DMA enabled

blazewicz, 2016-11-17:
Adding an issue as discussed in #2408 

For example, `tests/pyb/i2c.py` fails if following patch is applied:
```diff
diff --git a/tests/pyb/i2c.py b/tests/pyb/i2c.py
index a220f8e..0b3ba93 100644
--- a/tests/pyb/i2c.py
+++ b/tests/pyb/i2c.py
@@ -11,7 +11,7 @@ for bus in (-1, 0, 1, 2, 3, "X", "Y", "Z"):

 i2c = I2C(1)

-i2c.init(I2C.MASTER, baudrate=400000)
+i2c.init(I2C.MASTER, baudrate=400000, dma=True)
 print(i2c.scan())
 i2c.deinit()
```

yn386, 2022-08-31:
I met the same issue with NUCLEO-F446RE and I found some additional implementation is needed to i2c_ev_irq_handler.

I tried to fix in #9167.

dpgeorge, 2022-09-06:
Fixed by da50827657d6bc3024dd6acea76cb315d1cbbae1

closed 2022-09-06 by dpgeorge