micropython/micropython #2645: stmhal: STM32L4 boards don't build any more

dhylands, 2016-11-19:
With commit https://github.com/micropython/micropython/commit/8fa073364702e785226741012b5e94b6d2bf9674 the STM32L4 boards are broken

jamesbowman, 2016-11-21:
Yes, agreed. Trying:

    cd stmhal
    make BOARD=STM32L476DISC

At 7604de3 it builds cleanly. At 8fa0733 it fails with:

```i2c.c: In function 'i2c_ev_irq_handler':
i2c.c:343:23: error: 'I2C_TypeDef' has no member named 'SR1'
     if (hi2c->Instance->SR1 & I2C_FLAG_BTF && hi2c->State == HAL_I2C_STATE_BUSY_TX) {
                       ^
i2c.c:343:31: error: 'I2C_FLAG_BTF' undeclared (first use in this function)
     if (hi2c->Instance->SR1 & I2C_FLAG_BTF && hi2c->State == HAL_I2C_STATE_BUSY_TX) {
                               ^
```
etc, etc.

dpgeorge, 2016-11-21:
Thanks for the report.  Should be fixed by 21d82421cd7b3a3f74850cffaf84646ff49a04f2.

@blazewicz FYI.

closed 2016-11-21 by dpgeorge