micropython/micropython #2648: Benchmarking SDCard Write Speeds

r4d10n, 2016-11-22:
Trying to benchmark SDcard write speeds on STM32F7 Disco with DMA enabled... I found mentions of 3700 kbps somewhere, but I'm not getting even close ! For my application, I would be needing to write timestamped 24bits @ 200k.... 

> 190607 bytes in 6643 ms == 28.69291kBps

Not sure if the method I'm doing is right.. Code below:

Addendum: Is it possible to write to two SDcards simultaneously ? For redundancy.   

`import utime
import os	
import pyb

pyb.usb_mode('CDC+HID')

x = open('data-i.txt','w')

p = 0
t0 = utime.ticks_ms()

for i in range(0,10240):
	t = pyb.millis()
	p += x.write(str(t)+":"+str(os.urandom(3))+"\n")

t1 = utime.ticks_ms()
x.close()

total = utime.ticks_diff(t1,t0)

print(str(p) + " bytes in " + str(total) + " ms == " + str(p / total) + "kBps")

pyb.usb_mode('CDC+MSC')`

dhylands, 2016-11-22:
You should specify '/sd/data-i.txt' to make sure your file is being written to the sdcard and not to the internal filesystem.

Writing small amounts of data will be your worst case from a performance perspective. To get decent performance you'll want to be writing at least 512 bytes at a time and possibly more. Writing less than whatever the underlying flash block size (aligned on a flash block boundary) will result in your write becoming a read/write instead of just a write. As you increase the size of your writes, you'll typically see a sudden jump at some point. That jump usually corresponds to the underlying block size used by the flash. It might be even larger (like 4K or 16K).

You'll also want to use 'wb' mode instead of 'w' mode. 'w' mode uses strings and has to convert the strings to bytestrings. 'wb' uses bytestrings and doesn't require any extra conversion.

r4d10n, 2016-11-22:
Ahoy ! Just about everything mentioned up there rocked it clean... 

I tried writing 100 blocks (fixed preallocated string) with size multiples of 512, 4K, 16K as suggested. While testing 16K, I noticed a bump @ 32K, which I initially thought was some erroneous spike. 

> 1 : 1638400 bytes in 2310 ms == 709.2641kBps
> 2 : 3276800 bytes in 2884 ms == 1136.2kBps
> 3 : 4915200 bytes in 6146 ms == 799.7396kBps
> 4 : 6553600 bytes in 7725 ms == 848.3624kBps
> 5 : 8192000 bytes in 8540 ms == 959.2505kBps
> 6 : 9830400 bytes in 9307 ms == 1056.237kBps
> 7 : 11468800 bytes in 11205 ms == 1023.543kBps
> 8 : 13107200 bytes in 12528 ms == 1046.232kBps
> 9 : 14745600 bytes in 13827 ms == 1066.435kBps
> 10 : 16384000 bytes in 14511 ms == 1129.074kBps
> 11 : 18022400 bytes in 15175 ms == 1187.638kBps
> 12 : 19660800 bytes in 16837 ms == 1167.714kBps

But then as I tried it over again, there was visible increment.  So I averaged 1000 writes @ 32K and it gave me this :

> 2 : 32768000 bytes in 25614 ms == 1279.3kBps

once again: 

> 2 : 32768000 bytes in 25689 ms == 1275.566kBps

Almost 10Mbps !! I have verified the written file on disk and it looks good for me...

I happen to see the SDCARD_BLOCK_SIZE define inside sdcard.h. Altering that to 32K did give me some good steady and bit improved rates, but the USB-MSC failed to mount with dmesg saying  - Unsupported sector size 32768 :)

> 1 : 1638400 bytes in 1697 ms == 965.4684kBps
> 2 : 3276800 bytes in 3020 ms == 1085.033kBps
> 3 : 4915200 bytes in 4660 ms == 1054.764kBps
> 4 : 6553600 bytes in 5694 ms == 1150.966kBps
> 5 : 8192000 bytes in 6055 ms == 1352.931kBps
> 6 : 9830400 bytes in 8047 ms == 1221.623kBps
> 7 : 11468800 bytes in 9179 ms == 1249.461kBps
> 8 : 13107200 bytes in 10741 ms == 1220.296kBps
> 9 : 14745600 bytes in 11409 ms == 1292.453kBps
> 10 : 16384000 bytes in 11612 ms == 1410.954kBps

Does anybody have comparison figures with PyBoard or others ?? Please do post some. The benchmark code I tried is attached herewith. 

[sdcard-write-benchmark.py.txt](https://github.com/micropython/micropython/files/607689/sdcard-write-benchmark.py.txt)

Also, can I wire an extra SDcard interface to the board and instantiate a redundant disk somehow ? 

Cheers

dpgeorge, 2016-11-23:
> Also, can I wire an extra SDcard interface to the board and instantiate a redundant disk somehow ?

Yes, you can do it over SPI.  See drivers/sdcard/sdcard.py.

bvernoux, 2016-12-10:
I have done a read benchmark on hydrabus (use STM32F405@168MHz) => hydrafw
See https://github.com/bvernoux/hydrafw command `sd test-perf`
> sd show
Card type              : SD 2.0
Manufacturer ID        : 9c
OEM/Application ID     : 'SO'
Product name           : 'USD00'
Product revision       : 0.6
Serial number          : 633b3a24
Manufacturing date     : 2016-04
Capacity               : 7546 MiB
Capacity class         : HC (High capacity)
TAAC                   : 1 ms
Data transfer rate     : 25 Mbit/s
Command classes        : 0 2 4 5 7 8 10
Read block length      : 512 bytes
Write block length     : 512 bytes
Partial block reads    : no
Partial block writes   : no
Read-write factor      : 4
Misaligned block writes: no
Misaligned block reads : no
Contents copied        : no
Erase sector size      : 128 bytes
Minimum erase size     : 512 bytes
Write-protect possible : no
write-protected        : no
Driver stage register  : no
Relative card address  : 1388

> sd test-perf
Aligned sequential reads:
0.5KiB blocks:   5886 sectors/s,  3.013 MB/s
  1KiB blocks:   7890 sectors/s,  4.039 MB/s
  2KiB blocks:  11928 sectors/s,  6.107 MB/s
  4KiB blocks:  15568 sectors/s,  7.970 MB/s
  8KiB blocks:  18512 sectors/s,  9.478 MB/s
 16KiB blocks:  20480 sectors/s, 10.485 MB/s
 32KiB blocks:  21632 sectors/s, 11.075 MB/s
 64KiB blocks:  22272 sectors/s, 11.403 MB/s
Unaligned sequential reads:
0.5KiB blocks:   5892 sectors/s,  3.016 MB/s
  1KiB blocks:   5910 sectors/s,  3.025 MB/s
  2KiB blocks:   5956 sectors/s,  3.049 MB/s
  4KiB blocks:   6048 sectors/s,  3.096 MB/s
  8KiB blocks:   5872 sectors/s,  3.006 MB/s
 16KiB blocks:   5728 sectors/s,  2.932 MB/s
 32KiB blocks:   5760 sectors/s,  2.949 MB/s
 64KiB blocks:   5760 sectors/s,  2.949 MB/s

See also this very old bench https://github.com/bvernoux/hydrafw/wiki/HydraBus-microSD-read-performance

r4d10n, 2016-12-27:
Just tested with a Samsung EVO 32GB Class 10 disk.. Testing bigger writes till fill now.. 

1 : 1638400 bytes in 441 ms == 3715.193kBps
2 : 3276800 bytes in 831 ms == 3943.201kBps
3 : 4915200 bytes in 1285 ms == 3825.058kBps
4 : 6553600 bytes in 1587 ms == 4129.552kBps
5 : 8192000 bytes in 2104 ms == 3893.536kBps
6 : 9830400 bytes in 2452 ms == 4009.136kBps
7 : 11468800 bytes in 2926 ms == 3919.617kBps
8 : 13107200 bytes in 3309 ms == 3961.076kBps
9 : 14745600 bytes in 3681 ms == 4005.868kBps
10 : 16384000 bytes in 4320 ms == 3792.593kBps
11 : 18022400 bytes in 4683 ms == 3848.473kBps
12 : 19660800 bytes in 5205 ms == 3777.291kBps

pfalcon, 2017-05-13:
No bug report here, no further progress. Please move the discussing to the forum, http://forum.micropython.org . Thanks.

closed 2017-05-13 by pfalcon