micropython/micropython #2655: Setting the flash disk name to the board name

rolandvs, 2016-11-25:
In cases where you have a board that is different from the PYB board it would be nice to give the ```/flash``` drive the name of the board. This could be implemented by adding the following:

In `mpconfigboard.h` set the name of the board and add the label:
```
#define MICROPY_HW_BOARD_NAME "CLICKER2"
#define MICROPY_HW_FLASH_LABEL	(1)	// set the label of the drive in stmhal/main.c when defined
```
Change this in ```stmhal/main.c```
```
// set label
// - patched to give flashdrive the name of the board
#if defined(MICROPY_HW_FLASH_LABEL)        
    f_setlabel("/flash/"MICROPY_HW_BOARD_NAME); 
#else
   f_setlabel("/flash/pybflash");
#endif
```

tannewt, 2016-11-28:
Note that the label limit is 11 characters I believe. You probably want to drop the /flash/ prefix.

dhylands, 2016-11-28:
The /flash in this case is used to identfy the volume that the label should be changed for.
So f_setlabel expects /roodir/LABEL

tannewt, 2016-11-29:
Ah I see. Ignore me then. :-)

prusnak, 2016-11-29:
Is there any practical reason for this or just cosmetics?

dhylands, 2016-11-29:
It acts as a kind of confirmation about what you have plugged in. I noticed that all of my newer boards use names like DIS_F476 for the stlink flash file system (not the MicroPython one). For the MicroPython one, you can also change the label

Under linux, you can install a package called mtools, and then you can set the label to an arbitrary value by doing the following:
```
echo mtools_skip_check=1 >> ~/.mtoolsrc
sudo umount /dev/sdc1
sudo mlabel -i /dev/sdc1 -s ::labelname
```
This assumes that you pyboard was mounted at /dev/sdc1. This assigned label will then survive until you  reformat the filesystem on the pyboard (or change the label again).

rolandvs, 2016-11-29:
One can debate about what label to use :-).  I have it in the way I described, however (and the same goes for the generation of the initial files in the drive, see `stmhal/main.c`. I would like to set the label outside the `main.c` and use the board name or something else as suggested. I have a number of designs that all use the internal flash and not an sd card. It is just part of the configuration of the board solution.

dpgeorge, 2018-04-23:
The label of the internal flash filesystem is now configurable via the macro MICROPY_HW_FLASH_FS_LABEL

closed 2018-04-23 by dpgeorge