micropython/micropython #2656: pyb.bootloader() broken?

robert-hh, 2016-11-28:
Today I noticed, that pyb.bootloader() does not activate the bootloader any more. Actual version:
`MicroPython v1.8.6-106-g8103a66 on 2016-11-28; PYBv1.1 with STM32F405RG`
I know that I could use use like a week ago in the same set-up. That happens both on a pyboard 1.0 and pyboard 1.1. Activating the bootloader with the boot pin still works.

dhylands, 2016-11-28:
Hmm. The latest master seems to be v1.8.6-104-g0d75b0d and that seems to work (PYBV10 and PYBV11)

I tested with:
```
MicroPython v1.8.6-104-g0d75b0d on 2016-11-28; PYBv1.0 with STM32F405RG
```
and
```
MicroPython v1.8.6-104-g0d75b0d on 2016-11-28; PYBv1.1 with STM32F405RG
```

I couldn't find hash 8103a66 and 106 seems to indicate that its 2 commits newer than master?

robert-hh, 2016-11-28:
That's just a change I made in esp8266/Makefile to have the btree package as something that could be overridden. Nothing related to stmhal.
```
-MICROPY_PY_BTREE ?= 1
+MICROPY_PY_BTREE = 1
```

robert-hh, 2016-11-28:
Update: I tested that with an SD card in a larger board. Taking it out from that board did not change the picture, but removing the SD card. Then pyb.bootloader() worked. Funny enough, it also worked after I re-inserted the SD card (carefully powering off in between the SD card out-in cycles).

closed 2016-11-29 by robert-hh