micropython/micropython #2658: Timer.init() allocates memory (and thus can't be used in an ISR)

pjz, 2016-12-01:
Is it possible to allocate memory for the Timer object in the Timer constructor instead of in Timer.init()?   Otherwise it's impossible to start a time from a different ISR... which I'd like to be able to do.

dpgeorge, 2016-12-01:
I can't reproduce this problem.  Timer.init() doesn't seem to allocate any memory, and the following test runs fine on esp8266:
```python
import micropython, machine

def init(t, p, c):
    micropython.heap_lock()
    t.init(period=p, callback=c) # initialise the time while the heap is locked
    micropython.heap_unlock()

t = machine.Timer(-1)
init(t, 2000, lambda t:print(t))
```

pjz, 2016-12-01:
try this, then press the flash button.

```python
 from machine import Pin, Timer
 import micropython
 micropython.alloc_emergency_exception_buf(128)
 
 class Button:
     def __init__(self, pin):
         self.timer = Timer(-1)
         button = Pin(pin, Pin.IN)
         button.irq(trigger=Pin.IRQ_FALLING, handler=self.on_press)
 
     def on_press(self, newstate):
         print("start")
         self.timer.init(period=50, mode=Timer.PERIODIC, callback=self.show_end)
 
     def show_end(self, *args):
         print("end")

 Button(0)
```

dpgeorge, 2016-12-01:
The memory allocation is in the creation of the `self.show_end` bound-method (kind of like a closure, closing over the self variable).  To work around this, pre-create the bound-method object (the closure) in the Button constructor, like this: `self.bound_show_end = self.show_end`.  Then use `callback=self.bound_show_end` in the timer init method.  Note that `bound_show_end` is simply a variable and accessing it won't allocate any memory.

peterhinch, 2016-12-01:
Subtle!

pjz, 2016-12-01:
>Subtle!

Very! Thanks for the explanation, @dpgeorge !   Is there a FAQ or troubleshooting guide this should maybe go in?  As @peterhinch said, it's a bit subtle!

peterhinch, 2016-12-01:
It's not exactly a FAQ - as far as I know it's the first time it's cropped up :) Assigning a callback in an interrupt service routine is unusual (although valid).

pjz, 2016-12-01:
yeah, the Timer API is a bit confusing; I expected usage more like:

```
  t = Timer(-1, period=50, mode=Timer.PERIODIC, callback=self.show_end)
  ...
  t.start()
```

closed 2016-12-01 by dpgeorge