micropython/micropython #2659: esp8266: WebREPL doesn't start on soft reboot

tdicola, 2016-12-02:
I noticed with the latest 1.8.6 release (using the release build from micropython.org) you can connect to the repl and run `import webrepl_setup` to configure the WebREPL and access it just fine.  However if you soft reset the board with a Ctrl-D the WebREPL does not start, it only starts on a hard reset.  

For example here's serial output from a hard reset, notice the WebREPL started:
```
>>> rll��|�l�|�l�b|����r�b�b��nn�lnn���bp�lrlrlp�n��lbn�|���b��nn�l��l`�nnl`nr���nr�lp�n�r�����bn�|ll�b��nn�l`�nn�lnr���nr��`p�n�r������r�bn�|�b��nn��l`�nn�lnr���n��lr��n��ll��|��rrnb��l�b�lb쌜����b�lrlll��|��rrnb��ll�pr�b�b쌜��b��lrl�l��|��rrnb��l���b�b쌜���b�lbl����n�r��n|�llll`��r�l�l�l`��r�l�l�l`��r�l���ll`rl��rl���b��b�bbr�rb��n�nn�l��l�l��ll������l�n����bll�rp���bl�brlrlr�n�����n����brb��b쎜���l��bbl��ln�prl�l���#4 ets_task(40100164, 3, 3fff8398, 4)
WebREPL daemon started on ws://192.168.4.1:8266
Started webrepl in normal mode
could not open file 'main.py' for reading

MicroPython v1.8.6-7-gefd0927 on 2016-11-10; ESP module with ESP8266
Type "help()" for more information.
```

And here's the output from pressing Ctrl-D to soft reset, notice the WebREPL doesn't start and it's confused that the setup hasn't been run (it has been run and a webrepl_cfg.py file is available):
```
>>> 
PYB: soft reboot
#5 ets_task(40100164, 3, 3fff8398, 4)
WebREPL is not configured, run 'import webrepl_setup'
could not open file 'main.py' for reading
MicroPython v1.8.6-7-gefd0927 on 2016-11-10; ESP module with ESP8266
Type "help()" for more information.
>>> 
```

tdicola, 2016-12-02:
Also if it helps, using the `machine.reset()` API it does appear that the WebREPL restarts successfully:
```
>>> import machine
>>> machine.reset()
>>> 
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 32024, room 16 
tail 8
chksum 0x37
load 0x3ffe8000, len 1092, room 0 
tail 4
chksum 0xce
load 0x3ffe8450, len 3000, room 4 
tail 4
chksum 0x16
csum 0x16
l��|��rrnb��l�b�lb쌜����b�lrlll��|��rrnb��ll�pr�b�b쌜��b��lrl�l��|��rrnb��l���b�b쌜���b�lbl����n�r��n|�llll`��r�l�l�l`��r�l�l�l`��r�l���ll`rl��rl���b��b�bbr�rb��n�nn�l��l�l��ll������l�n����bll�rp���bl�brlrlr�n�����n����brb��b쎜���l��bbl��ln�prl�l��#4 ets_task(40100164, 3, 3fff8398, 4)
WebREPL daemon started on ws://192.168.4.1:8266
Started webrepl in normal mode
could not open file 'main.py' for reading

MicroPython v1.8.6-7-gefd0927 on 2016-11-10; ESP module with ESP8266
Type "help()" for more information.
>>> 
```

dpgeorge, 2016-12-05:
Soft reset on ESP8266 is not well supported for resetting anything network related.  But actually the cause of the problem here is most likely that importing a module a second time after it already failed (in this case the webrepl_cfg module) does not work correctly, see #967.

dpgeorge, 2016-12-05:
Looking more deeply into this, it seems that modules/webrepl.py catches all exceptions when trying to start webrepl in normal mode, even if the exception is not related to webrepl not being set up.  In particular it can catch OSError from running out of memory trying to make the sockets, and will report that as webrepl not configured.

It's suggested not to use soft reset on esp8266 when using any network features.

fede2cr, 2017-02-24:
Actually, if you use WebREPL to control a robot without USB cable, is normal to upload files vía Web and then you need a way to re-load those files.
Ctrl+D kills the WebREPL so you have to walk over to the robot and push the reset button on it.
This makes the workflow that on an SAM21 would be really simple, on an ESP8266 you have to walk over to the robot and press the hard reset button.

Should the ESP8266 disable soft_reset after the network funtions are initialized? Is it possible to force a hard_reset insted?

dpgeorge, 2017-02-27:
> Is it possible to force a hard_reset insted?

Yes, do machine.reset().

dpgeorge, 2023-06-05:
Soft reset has been improved a lot due to `gc_sweep_all()`.  So this may be fixed.

I'll close this due to it being a very old issue that has had no other updates.

closed 2023-06-05 by dpgeorge