micropython/micropython #2660: UART write timeout clarification

lorenzschmid, 2016-12-04:
Currently using micropython for a [small project](https://github.com/lorenzschmid/pyRN2483) for my micropython board I tried to figure out how I can set the write timeout mentioned in the [`UART.write`](https://docs.micropython.org/en/latest/pyboard/library/pyb.UART.html#pyb.UART.write) command documentation which states:

> Write the buffer of bytes to the bus. ... If a timeout occurs and no bytes were written returns `None`.

Looking at the source in [stmhal/uart.c](https://github.com/micropython/micropython/blob/master/stmhal/uart.c) I found two timeout parameters ([line 88/89](https://github.com/micropython/micropython/blob/1779a62085b287f10a02843feabc07d431671227/stmhal/uart.c#L88)) which seem to specify timeouts for receptions only (and not transmission/write):

    struct _pyb_uart_obj_t {
        ...
        uint16_t timeout;                   // timeout waiting for first char
        uint16_t timeout_char;              // timeout waiting between chars

Nevertheless looking at the transmitting function [`pyb_uart_writechar()`](https://github.com/micropython/micropython/blob/1779a62085b287f10a02843feabc07d431671227/stmhal/uart.c#L744) one can see that the same timeout is applied for transmission as well (incomplete excerpt):

    STATIC mp_obj_t pyb_uart_writechar(mp_obj_t self_in, mp_obj_t char_in) {
        pyb_uart_obj_t *self = self_in;

        // write the character
        if (uart_tx_wait(self, self->timeout)) {
            status = uart_tx_data(self, (uint8_t*)&data, 1);

This leads to two questions:

- Did I understand this correctly meaning `uint16_t timeout` is applied to reception and transmission?
- If yes, wouldn't it be simpler to use two timeouts, one for transmission and one for reception (whereby `uint16_t timeout_char` would be merged with the reception timeout)? This is also what is done in the [pyserial library](http://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial) having the parameters `timeout` and `write_ timeout`.

dhylands, 2016-12-04:
The only way that you can get a timeout on writes is if you're using flow control and the other end has said that it's busy, or you specify a timeout which is less than a character time. If you're not using flow control, then the timeout doesn't really do anything on writes.

dpgeorge, 2016-12-04:
> Did I understand this correctly meaning uint16_t timeout is applied to reception and transmission?

Both the `timeout` and `timeout_char` parameters are used for both RX and TX.  `timeout` is used as the timeout for sending/receiving the first char in a call (eg `read`, `write`, `readchar`, `writechar` methods), and `timeout_char` is used as the inter-character timeout for reads/writes that transfer multiple characters.

> If yes, wouldn't it be simpler to use two timeouts, one for transmission and one for reception (whereby uint16_t timeout_char would be merged with the reception timeout)? This is also what is done in the pyserial library having the parameters timeout and write_ timeout.

You would need 4 timeouts, 2 each for RX and TX.  I think it's simpler to just have the 2 because the required timing for RX and TX should be very similar (since they both use the same configured baudrate).  If you really need different timeouts for RX and TX you could reconfigure the timeouts before read or write calls.

lorenzschmid, 2016-12-11:
Thanks for your answers and explanations and sorry for my late reply: I wanted to reflect it directly in the documentation and had only now the time to create a corresponding PR.

Concerning the amount of timeouts, it is less clear to me why one would need to differentiate between first communicated char and inter-char-communication. But I might think not enough on a hardware level there and I definitely see your point considering the Baud-rate.

In any case I think this issue is solved. Thanks again!

closed 2016-12-11 by lorenzschmid