micropython/micropython #2661: Clearer error messages for networking on ESP8266

askvictor, 2016-12-04:
I'm occasionally getting OSError -2 when a networking problem occurs. It would be considerably more useful if this error described, at the very least, that this was a network error, or even better, what type of network error.

dpgeorge, 2016-12-04:
> I'm occasionally getting OSError -2 when a networking problem occurs.

This is probably a bug, since OSError error numbers should be positive and should in most cases refer to a known errno and give you a string describing the error (eg OSError: ECONNRESET).

Can you please give more detail as to when you get this error, after calling which internal function, eg socket.recv?

askvictor, 2016-12-05:
So I can re-create the error by using upip. If I reboot the device, network stuff seems to work fine, then I'll try to install a package with upip (I was trying to install umqtt.simple). This returns a 404, and thereafter any attempt to access the network results in OSError: -2; for example a call to `socket.getaddrinfo('micropython.org', 80)[0][-1]` does this. After a reset things work fine. It appears to only be a problem if the upip install returned a 404 - if the package installed successfully it seems to continue fine, and a couple of packages gave back 301 and things continued to work.

dpgeorge, 2016-12-05:
Ok, that looks like it could be an internal lwip error number, which is ERR_BUF, which can be mapped to ENOBUFS.  Or it could be that the lwip_getaddrinfo_cb function is erroring out.

Do you get the "-2" error from anything other than `getaddrinfo`?

pfalcon, 2016-12-05:
getaddrinfo() has own own error namespace (not within errno.h). In CPython, it's mapped to socket.gaierror exception. We just map it (inconsistently across ports) to OSError values outside positive integers domains as used for normal OSError's.

askvictor, 2016-12-05:
Looks like it is just getaddrinfo() that is failing; subsequent calls to upip.install() after a 404 end up failing when it hits getaddrinfo().

I guess there is the issue of the error message, and another of the getaddrinfo() failing after a 404 from upip.install()

askvictor, 2016-12-07:
I've just reproduced this on another device; am going to create a seperate issue for the 404->OSError issue

DSIW, 2017-01-03:
I'm using the ESP8266 on the Wifi Witty board.

I got this error after 5 requests to the same host. After I added `socket.close()` after sending the bytes, it worked and I didn't get the OSError -2 from getaddrinfo().

I added this line to the documentation via PR #2756.

dpgeorge, 2020-06-09:
I'll close this issue due to lack of discussion/progress.

closed 2020-06-09 by dpgeorge