micropython/micropython #2665: RFC Installing Python packages to platforms lacking a network interface

peterhinch, 2016-12-06:
I raised this on the forum. The suggested solution was to use upip to install the package to the Unix build, then copy the directory tree from ~/.micropython/lib to the device.

While this works, it seems inelegant. It's also unfriendly to users of Windows and OSX. Is there a platform independent way to do this? If not, any thoughts on the design of such a solution.

dpgeorge, 2016-12-07:
> The suggested solution was to use upip to install the package to the Unix build, then copy the directory tree from ~/.micropython/lib to the device.

You could also mount the device and use the "-p" option to upip to specify the target directory as the mounted filesystem.

> Is there a platform independent way to do this?

The unix version should run on OSX, and the windows version on windows.  You should also be able to use CPython's pip to install the files (to a local dir) and then copy them over.  Apart from that I don't know an easier way.

peterhinch, 2016-12-07:
Thanks for that. I've posted the above in the forum.

closed 2016-12-07 by peterhinch