micropython/micropython #2667: PyBoard`s SD card dies on Ctrl+D in REPL

kamikaze, 2016-12-08:
When SD card is being inserted and code (that initializes SPI and sends some init data to device) launched from it in REPL and then Ctrl+D is being pressed - BOOM, SD card error occurs.

Here is the sample initialization: https://github.com/kamikaze/pyboard-examples/blob/master/display/__init__.py#L18

dpgeorge, 2016-12-08:
Regarding the ETIMEDOUT error for SPI.write(): yes there is a hard-coded timeout of 100ms for SPI transfers (for machine.SPI, not for pyb.SPI), and it looks like you hit that limit with 256*64 bytes sent out at 1mbps.  That will need to be fixed in the SPI driver (eg set the timeout based on the number of bytes being transferred).

But you did get around that by increasing the baudrate to 10mbps.  Then the main issue (reported in this ticked) is the SD card not working after a soft reset.

I tried to reproduce this error but could not.  Here is my set-up and steps:
- PYBv1.1 running v1.8.6-155-g5e39300
- SD card inserted
- download latest code and put it in root of SD card: https://github.com/kamikaze/pyboard-examples commit 2413fbe7888daac1a9d34e40d2a030beead28af0
- boot the pyboard and execute at the repl the following: `import disp`, `d=disp.create_display()`, `d.send_buffer()`.
- press ctrl-D to do a soft reset
- SD card mounts correctly and I can `import disp` again and do the test again

Note that I don't have any display connected.  Maybe the problem is only there if the display is attached?

kamikaze, 2016-12-08:
yes, it occurs when display is connected. you don't even need to send_buffer(). Will test w/o display connected later.

kamikaze, 2016-12-08:
with display disconnected this error doesn't occur. As well as if you will comment CS pin initialization line ) so it only fails with display attached.

kamikaze, 2016-12-10:
I can't even read files with open() and f.read() when display is initialized. But I can read before init. I will provide a full PoC code that works from internal memory and doesn't from SD

kamikaze, 2016-12-11:
So here I am again.

To reproduce both ways:

1) get "display" and "frames" directories, and "animation.py" file from https://github.com/kamikaze/pyboard-examples
2) store "frames" directory on flash and SD card
3) store "display" directory and "animation.py" on flash (actually it doesn't matter where you will run the code from)
4) Run in REPL and get the output:

`
>>> import animation
>>> animation.run_test_from_flash()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/flash/animation.py", line 24, in run_test_from_flash
  File "/flash/animation.py", line 21, in run_test
  File "/flash/animation.py", line 16, in run_animation
  File "/flash/display/ssd1322.py", line 95, in send_buffer
  File "/flash/display/common.py", line 38, in write
KeyboardInterrupt: 
>>> animation.run_test_from_sd()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/flash/animation.py", line 28, in run_test_from_sd
  File "/flash/animation.py", line 21, in run_test
  File "/flash/animation.py", line 13, in run_animation
  File "/flash/animation.py", line 7, in read_frames
OSError: [Errno 5] EIO

(First test is being killed manually by hitting Ctrl+C)
`

kamikaze, 2016-12-12:
I've found the reason (but still not a solution):

https://github.com/kamikaze/pyboard-examples/blob/master/display/ssd1322.py#L64

This line turns off a sleep mode of the screen. Datasheet says that in sleep mode display consumes 10 uA. So I suppose when sleep mode turns off (display starts to show stuff only then) - more current is being consumed and something goes wrong with SD. Is it possible?

dpgeorge, 2016-12-13:
> So I suppose when sleep mode turns off (display starts to show stuff only then) - more current is being consumed and something goes wrong with SD. Is it possible?

Yes, excessive current draw is most likely the problem.  I see you're using a 256x64 OLED display, so that's 16k pixels.  If each draws around 10uA average (conservative estimate) then that's 160mA total.  PYBv1.1's LDO can source 250mA of current, but the MCU will use about 60mA if you are doing lots of processing.  So you are definitely on the edge of using too much power for the LDO.

I would suggest to measure the current consumption of the device, then you'll know for sure.  And then try powering it with an external power supply, and make sure to include a few decoupling capacitors.

dpgeorge, 2017-01-17:
Assuming that this was an issue with the power supply, not the software.

closed 2017-01-17 by dpgeorge