micropython/micropython #2668: ESP8266 : set up a new mac adress : ValueError: invalid buffer length

nithir, 2016-12-08:
Hello, 

When i try to set a new MAC adress on the esp8266 i get error.

i try this code:

```
>>> import network;
>>> new_mac = network.WLAN().config('mac')                
>>> network.WLAN().config(mac=new_mac)                
Traceback (most recent call last):                                                        
  File "<stdin>", line 1, in <module>                                                     
ValueError: invalid buffer length  
```
The current mac is this one :
```
>>> network.WLAN().config('mac')                                                          
b'\xa0 \xa6\n\xcd\xbb'
```

I find the related source code here : esp8266/modnetwork.c
from line 302 to line 311

```
case QS(MP_QSTR_mac): {
    mp_buffer_info_t bufinfo;
    mp_get_buffer_raise(kwargs->table[i].value, &bufinfo, MP_BUFFER_READ);
    if (bufinfo.len != 6) {
        nlr_raise(mp_obj_new_exception_msg(&mp_type_ValueError,
            "invalid buffer length"));
    }
    wifi_set_macaddr(self->if_id, bufinfo.buf);
    break;
}
```

Obviously the code test if buffer length is different from 6 but the actual mac have a length of 6...

```
>>> len(b'\xa0 \xa6\n\xcd\xbb')                                                           
6 
```  

So i am unable to set a different mac adress on my board... :(

Maybe i do something wrong... please help me ?

dpgeorge, 2016-12-08:
Make sure you do `wlan = network.WLAN(network.STA_IF)` to get the station interface, then make sure it's active, `wlan.active(True)`, then try to set the mac.

nithir, 2016-12-08:
Thanks for answer.

 I've tried this

```
>>> import network
>>> sta_if = network.WLAN(network.STA_IF)
>>> sta_if.active(True)
>>> sta_if.active()
True
>>> sta_if.config(mac=sta_if.config('mac'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid buffer length
>>> sta_if.config('mac')
b'\xa0 \xa6\n\xcd\xbb'
```

dpgeorge, 2016-12-08:
I think you have an old version of the firmware.  There was a bug in the mac setting code which was fixed on 31st Oct 2016, e377f3cb40b34a562ff71753c112edec3d1bd4d1.  Please upgrade to v1.8.6.

nithir, 2016-12-08:
I'm sorry but v1.8.6 doesn't seems to work on my "LoLin v3 nodemcu" chinese clone.
No problème to load the firmware. But there is no way to access to REPL.
The only things i see is the led flashing quickly a great number of time...

With v1.8.5 almost everything (except the mac problem) works fine.

nithir, 2016-12-08:
Well, after finding a trick on the Internet about connecting **GPIO0 to Ground and pushing reset button**... 
I've uploaded to my board **v1.8.6** and it works fine now...

so i retried.
```
>>> import network
>>> sta_if = network.WLAN(network.STA_IF)
>>> sta_if.active(True)
#5 ets_task(4020ed88, 28, 3fff9f68, 10)
>>> sta_if.active()
True
>>> sta_if.config(mac=sta_if.config('mac'))
>>> sta_if.config('mac')
b'\xa0 \xa6\n\xcd\xbb'
>>> 
```

and finally **it works**.

Thank you for the time spend to help me

closed 2016-12-09 by dpgeorge