micropython/micropython #2669: Byte order setting in from_bytes/to_bytes

Snigelson, 2016-12-08:
Discovered this on a big endian MIPS machine, while trying to use the re library.

The following, though, is using the latest Micropython on a little endian machine.

    MicroPython v1.8.6-155-g5e39300 on 2016-12-08; linux version
    Use Ctrl-D to exit, Ctrl-E for paste mode
    >>> int.from_bytes(b'\x00\x00\x00\x01', 'little')
    16777216
    >>> int.from_bytes(b'\x00\x00\x00\x01', 'big')
    16777216

The `byteorder` parameter is completely ignored. The third parameter, `signed`, cannot even be passed.

Snigelson, 2016-12-09:
Note that  everything works fine when byteswapping the integer using a struct.

pfalcon, 2016-12-09:
Right. Sorry, but I forgot the situation myself :-I. This would be the duplicate of https://github.com/micropython/micropython/issues/682 then. And it's marked as "prio-low", exactly because there're 2 other ways to do it: via struct and via array modules.

pfalcon, 2016-12-09:
Ok so what could have been done here is making sure that using unimplemented "big" byteorder doesn't silently produce incorrect result and instead throws exception: https://github.com/micropython/micropython/commit/9d787de2a17f99cfeb715507857b92d54b023715 . The rest is #682, https://github.com/micropython/micropython-lib/issues/128

closed 2016-12-09 by pfalcon

Snigelson, 2016-12-09:
Is there any clean way to do it conditionally, i.e. byte swapping only if running on big endian?

peterhinch, 2016-12-10:
Perhaps best handled in the application. Device drivers do byte swapping where the aim is to match the endian-ness of the hardware device; this may not match that of the MPU.