micropython/micropython #2670: Firmata Support - Feature? Thoughts?

cefn, 2016-12-08:
I'm trying to figure out the steps to get a Firmata client running within Micropython on an ESP8266 platform. For example, this would enable the ESP01 devices to be used as a python interpreter to drive an ATMEGA328 running [StandardFirmata](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino), giving robust IO and providing access to pretty much anything for which Arduino libraries exist (plus you could deploy a modified Firmata firmware to the Arduino to use Sysex messages if there was something unsupported by established Firmata protocol extensions).

There are quite a few candidate python client implementations 
* Pymata Asyncio https://github.com/MrYsLab/pymata-aio/tree/master/pymata_aio - very up-to-date - Firmata2.5 
* PyFirmata https://github.com/tino/pyFirmata - Outdated Firmata version compatibility (Firmata 2.1), but my favourite Python Firmata library until I saw pymata

I am unsure whether any of these can be used within Micropython, or if there's a fundamental issue with porting them to run on Micropython (using the serial API for communication with the Arduino).

Can anyone provide commentary on the compatibility issues I need to consider when trying to get the goodness from a stable python library (typically those reliant on pySerial) across to run on an ESP8266 Micropython image.

For many cases, I suppose I can roll my own Serial protocol and write my own ATMEGA firmware, but Firmata could provide a really good pre-designed protocol, and the potential for porting established python client implementations is really exciting.

Any thoughts on obstacles to porting these? If there are obvious advantages/disadvantages to porting any of the different client libraries, that micropython experts can anticipate, it would probably save a world of pain from me naively copy-pasting things which are never going to work.

Neon22, 2016-12-08:
I started one but abandoned it here:
https://github.com/Neon22/micropython-firmata
This discussion may be better on the forum though...

cefn, 2016-12-08:
I saw the micropython-firmata project, but I understood it to be a firmata implementation not a firmata client (to control a firmata implementation). Did I get it backwards?

Neon22, 2016-12-08:
yes - you're right - sorry

cefn, 2016-12-09:
Looks like my favoured candidate PyMata cannot be trivially ported because it relies on collections.deque for the serial transport abstraction...

https://github.com/MrYsLab/PyMata/blob/master/PyMata/pymata.py

https://github.com/micropython/micropython/wiki/Differences

Neon22, 2016-12-09:
deque only appears to be used for its proficiency at fast push, pops. You could try just using a list and managing it yourself.
https://docs.python.org/2/library/collections.html

utzig, 2016-12-10:
Seems to be just a producer/consumer. You can use a `list()` instead, just change the declaration at:

https://github.com/MrYsLab/PyMata/blob/master/PyMata/pymata.py#L52

to:

`command_deque = list()` and remove `from collections import deque`.

inside `pymata_command_handler.py` change the `popleft()` calls from `self.pymata.command_deque.popleft()` to `self.pymata.command_deque.pop(0)`.

It will extremely inefficient but it should work!

Wei1234c, 2017-01-20:
I use MicroPython with ESP8266 to build a Hub-Workers system. 

You can read/write GPIO remotely, or call remote function and get result returned.

Also, code and script file can be deployed over the network and executed dynamically.

I don't have document in English, sorry. However, I think Google Translation will help, or you can read the code directly.

Hope it helps. 
 
https://github.com/Wei1234c/Plastic_Network_of_Things_with_MicroPython

mhoffma, 2017-01-21:
Did you make progress I was thinking about the same thing and just came across you're activity?

Wei1234c, 2017-01-21:
Just wrote a introduction in English.
http://wei1234c.blogspot.tw/2017/01/elastic-network-of-things-with.html

Github repository had been moved to
github.com/Wei1234c/Elastic_Network_of_Things_with_MicroPython

cefn, 2017-01-22:
@mhoffma no, I haven't started trying to port the pymata code, although it looks promising from a naive point of view, I keep wondering whether there are differences between python and micropython that I haven't yet understood and will only discover by trying it. Also I haven't quite fathomed whether the asyncio variant is worth exploring, ( https://github.com/MrYsLab/pymata-aio ) or whether the protocol engine needs pulling out and using on its own with events marshalled and unmarshalled explicitly using micropython libraries to minimise porting complexity.

mhoffma, 2017-01-23:
I started this project as a place to dig into micropython as I want a nice platform for the robots my 10 year old son is working on, I pulled a copy of of someone simple pyduino that didn't work at all and started to make it work, the code can read and write the digital pins and receive the analog updates atm. my setup is a ESP01 --> mini pro with a mcp1700 5v to 3.3v power converter.  I use gpio2 to reset the pro mini, and connect TX->RX and RX->TX between the tow modules. I have a small update to micorpython that disables the use of the uart still need to figure out how to get that into the mainline of micropython.  But this scheme does work like a charm.

https://github.com/mhoffma/esp/

I plan to figure out how to get this to work in async fashion and get my micro python patch considered for inclusion into the mainline. I probably could use help on my api function name that is used to disable the stdio function over uart.

pfalcon, 2017-04-05:
Closing due to inactivity. Please use this tracker to report bugs in the core MicroPython project. Please use the forum, http://forum.micropython.org/ , for other discussions.

closed 2017-04-05 by pfalcon

cefn, 2017-04-29:
Moved discussion to https://forum.micropython.org/viewtopic.php?f=16&t=3329