micropython/micropython #2672: RFC: esp8266, setting location for native code generation

dpgeorge, 2016-12-09:
One of the stretch goals of the ESP8266 Kickstarter was to implement the native emitter (`@micropython.native`, `@micropython.viper`) and inline assembler (`@micropython.asm_xtensa`).  This is now pretty much complete, see 45a6156dfd327f14f58c30ebf9159a92a5567ed5 and f76b1bfa9f59fcfa03837c6934ad51d2db3ff4a3.

A notable point with the esp8266 is that you can't execute machine code from the heap, from dram (it's data-ram).  There are 2 locations where code can be written to and executed from:
- iram1 region
- flash, below 1Mbyte

The iram1 region (32k total) actually has a little bit of room at the end (about 720 bytes currently) and this can be used to put code generated by the native emitter.  But it's not much space and to get the most of the native emitter one would need to write the code to flash and execute it from there.

To specify the location for native code, I've implemented a (preliminary) function in the esp module:
```python
esp.set_native_code_location(start, len)
```
If you call this function with `None` for both parameters then native code is stored at the end of the iram1 region.  Otherwise you need to specify in `start` the byte offset within the flash (starting at 0, up to 1MByte) where native code can go.  And the `len` parameter is the maximum bytes of flash to use.  The flash will be erased as needed as native code is generated, and native code will be stored sequentially to the flash region specified.

The default behaviour (if you don't ever call `esp.set_native_code_location`) is to use the end of the iram1 region.  This means you can try out some simple native/viper/asm_xtensa functions without wearing out the flash with erase/write cycles.

To use some flash to store native code you will need to make sure the filesystem is not using that part of the flash, which means adjusting how much flash is used in modules/flashbdev.py.

The purpose of this issue is to discuss a way to deal with the problem of allocating flash space for native code, and whether the `esp.set_native_code_location` function is a good way of selecting the location.

pfalcon, 2016-12-09:
> The flash will be erased as needed as native code is generated, and native code will be stored sequentially to the flash region specified.

How allocation of this region is controlled? Is it just bump-allocated, with "free" pointer stored in RAM, so with a reboot, allocation starts from the beginning (and a reboot is also a way to "garbage-collect" that area)?

pfalcon, 2016-12-09:
I had very far-fetched idea to develop a simple "extent-based" file system for the 1st megabyte of flash, which would allow to "support" mmap() operation on it (e.g. to execute bytecode and native code directly out of it).

dpgeorge, 2016-12-09:
> How allocation of this region is controlled? Is it just bump-allocated, with "free" pointer stored in RAM, so with a reboot, allocation starts from the beginning (and a reboot is also a way to "garbage-collect" that area)?

Yes, exactly.  So you can't free parts of it, you must free it all at once (which you could do with a call to esp.set_native_code_location passing the same parameters as before).

And you necessarily need to free it all on a reboot, ie you can't reuse the generated machine code.  This is because qstr's are stored within the native data and qstr id's can change after a soft/hard reboot.

> I had very far-fetched idea to develop a simple "extent-based" file system for the 1st megabyte of flash, which would allow to "support" mmap() operation on it (e.g. to execute bytecode and native code directly out of it).

This would be useful to help reclaim some RAM.  It would also be good to have the feature usable on all ports.

pfalcon, 2016-12-13:
Ok, to move forward with this, I guess we need to reserve some low-ROM memory for native codegen, and document its location. I'd say 4K (one page) is enough, any other ideas?

dpgeorge, 2016-12-14:
> I'd say 4K (one page) is enough, any other ideas?

It depends on how much people will use this feature.  I guess we can start with 4k assuming native code will only be used for really critical stuff (ie not many functions) and increase (eg to 8k) it if there's demand/requests.

But anyway, users can always adjust the default settings by changing the _boot.py function if they need more room for native code.

The other thing to decide is whether native code goes in iram or flash with the default settings.  Having the default as iram means people can play with some simple native code (eg viper or inline-asm) without worry about wearing out the flash, and then put stuff in flash when they have finalised the code.

dpgeorge, 2017-01-04:
In the default config there is now a reserved sector for for native code (or other general purpose use).  See bae7798f1ef531c862421b37c0d5d462e34a6d2f and c3f70c603efcb053810dbee8d5cc9f23ed62d01f.

pfalcon, 2017-04-05:
@dpgeorge : Should we close this now, or it's still super-provisional and waits for more feedback?

dpgeorge, 2017-04-05:
I guess it can be closed, it's been long enough for feedback.  And the API might even work on esp32, if that ever gets native support.

closed 2017-04-05 by dpgeorge