micropython/micropython #2673: Adafruit's Extended Ascii font as stmhal/framebuf's default font.

Cediddi, 2016-12-11:
Hello, 
Lately I was curious about if I could replace framebuf's default font with [Adafruit's 5x8 font](https://github.com/adafruit/micropython-adafruit-bitmap-font). Current 8x8 font is nicely packed and readable but misses extended ascii characters. Size difference is %20. Current font is 1024 bytes, extended font is 1275 bytes long. What are your thoughts about replacing the default font or adding option to use a user given font, rather than default font.

mcauser, 2016-12-11:
The 5x8 font is nicer on smaller displays, such as my SSD1306 based 128x32 and 64x48 OLEDs. The default "pet me" 8x8 font looks very bold in comparison.

Rather than replacing font A for font B, I would find it much more useful to be able to load custom fonts at runtime or set the default before compiling (so I don't have to store multiple).

The TomThumb font is only 3x5, for your tiny displays, or where you want a stack of text:
https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Fonts/TomThumb.h

It would be nice if there were a tool for creating custom fonts from .ttf files, such as this Adafruit one: https://github.com/adafruit/Adafruit-GFX-Library/tree/master/fontconvert

peterhinch, 2016-12-11:
You might like to look at [font_to_py](https://github.com/peterhinch/micropython-font-to-py.git) which provides a way to convert arbitrary ``ttf`` and ``otf`` fonts to Python sourcecode. This can be frozen as bytecode if necessary. [This collection of code ideas](https://github.com/peterhinch/micropython-samples.git) has some sample code for using such fonts with the SSD1306.

Cediddi, 2016-12-11:
I also use 64x48 oled. My main issue was about extended ascii characters. TomThumb font looks nice but I rather have a fixed width font. For such a small screen, a tailor made font would be the best in my opinion. On the other hand, your SSD1306 class might save my day with a little bit tweaking.

dpgeorge, 2016-12-14:
> What are your thoughts about replacing the default font or adding option to use a user given font, rather than default font.

There should be a way for the user to specify their own font, then they can use what's most suitable for their application/display.  And it also overcomes the problem with font licenses.

So there needs to be a definition of a font binary format (eg first 2 bytes specify fixed w/h, next 2 bytes specify first/last character, rest of bytes are data), as well as a way to specify the font data to use (eg framebuf.set_font(buf)).

wangshujun-tj, 2024-04-30:
我在做中文支持的时候增加了字体部分
https://github.com/wangshujun-tj/mpy-Framebuf-boost-code
欢迎直接引用

jonnor, 2024-09-29:
There are many solutions available for MicroPython for using custom fonts, which allows to render those fonts to a standard framebuffer. Examples include:

- https://github.com/peterhinch/micropython-font-to-py
- https://github.com/antirez/microfont

So I think we can close this issue?

robert-hh, 2024-09-29:
Agreed.

closed 2024-09-29 by jonnor