micropython/micropython #2674: Older releases as a build firmwares

iBobik, 2016-12-11:
Currently [Releases tab](https://github.com/micropython/micropython/releases) on GitHub contains only ZIP archives of the source code, not firmwares for a boards. [Website](http://micropython.org/download) contains only some last versions.

I am looking for an old release for ESP8266, so I tried this link (because it is not listed on the website), but it is not working:
http://micropython.org/resources/firmware/esp8266-20160201-v1.6.bin

I propose to upload all builds to releases here on the GitHub, because it seems clean, straightforward and GH provides free hosting for it. But also archive on the website helps.

My motivation: Current versions bricked my board, so I want to try some older version, because it worked in the past.

robert-hh, 2016-12-12:
You can always build any release from the git repository. It 'only' requires to install the build environment, which is best done on a Linux system.

iBobik, 2016-12-12:
I understand I can build it. I tried, but it is too hard core for one use
for me. :-)

Will be much helpful if you provide whole environment as a Docker image.
And this image can be reused also in travis builds. If I try to do it like
this, will you accept PR?

Jan Pobořil

robert-hh, 2016-12-12:
After writing, I tried to build that, but it does not match with the recent version of the ESP-open-sdk. Therefore I deleted my comment. I have to go back to the 1.5.4 version of the sdk.

iBobik, 2016-12-12:
That is the reason why archive is needed. :-)

dpgeorge, 2016-12-13:
> I am looking for an old release for ESP8266, so I tried this link (because it is not listed on the website), but it is not working:
> http://micropython.org/resources/firmware/esp8266-20160201-v1.6.bin

Note that esp8266 builds were not automated until v1.8, so that's why it doesn't exist.

> I propose to upload all builds to releases here on the GitHub, because it seems clean, straightforward and GH provides free hosting for it. But also archive on the website helps.

If we did this, the uploads to github would just be a mirror/copy of the firmware on micropython.org/download.  The latter is automated, and is built only *after* a github release is made.

> Current versions bricked my board, so I want to try some older version, because it worked in the past.

Going back to v1.8 should be enough.  You could also try erasing the entire flash and then using the latest esptool.py to upload the most recent release.

iBobik, 2016-12-13:
> Current versions bricked my board, so I want to try some older version, because it worked in the past.
> 
> Going back to v1.8 should be enough. You could also try erasing the entire flash and then using the latest esptool.py to upload the most recent release.
> 
Tried erase_flash and v1.8 on two devices with the latest esptool.py, but device just starts AP (possible to connect to Wi-Fi, but not to WebREPL) and serial REPL is not responding.

Should I create another issue for it or forum post?

dpgeorge, 2016-12-13:
> Should I create another issue for it or forum post?

Please try to debug it yourself first using hints from the forum, eg read the "Announcements" section in this forum: http://forum.micropython.org/viewforum.php?f=16.

If you come across a reproducible bug then please post here.  Otherwise please ask for general usage help at the forum, thanks!

iBobik, 2016-12-14:
@dpgeorge 
> Note that esp8266 builds were not automated until v1.8, so that's why it doesn't exist.

Where are build scripts? I see .travis.yml in the root, but it does not seems like building esp8266.

If I make Dockerfile containing complete build environment and add it to .travis.yml (replace before_script), will PR be accepted?

iBobik, 2016-12-14:
Aha, I see Travis does not support using Docker image as a environment for builds like GitLab CI.

But it still is needed to have build automation available to everybody.

dpgeorge, 2016-12-15:
> But it still is needed to have build automation available to everybody.

Well, you can just type "make" within the esp8266 directory and it builds the code automatically.  And this is exactly how the automated build system works to produce the firmware at micropython.org/download.  There is no dockefile or anything like that.

iBobik, 2016-12-15:
It probably is not everything. At first it requires open-esp-sdk, this has
some dependencies, make of the MicroPython has some dependencies. And
really, it took me half hour until I found that not all dependencies were
met and I do not know what of them specifically.

Build machine have some environment where it works, but it is not easy for
everybody to have the same. And you have no guarantee that every build will
be consistent, because you cannot be sure what dependencies version will be
used.

This is place for declared build environment like Docker :-)

Jan Pobořil

dpgeorge, 2016-12-15:
> Where are build scripts? I see .travis.yml in the root, but it does not seems like building esp8266.

I just remembered that there is actually a branch of this repo, travis-esp8266, which does build the esp8266 port on Travis.  You can checkout that branch to see how the .travis.yml file looks, maybe that will help you build it.

Otherwise, the instructions in esp8266/README.md should be sufficient to get a build system working.  And in fact should be able to make a consistent and reproducible build environment, at least as much as that is possible (the toolchain is a fixed-version, fixed by esp-open-sdk).

iBobik, 2016-12-15:
Nice, thanks for a tip :-)

Jan Pobořil

pfalcon, 2017-05-14:
No further progress, closing. The project is busy with releasing new, not old versions of MicroPython.

closed 2017-05-14 by pfalcon