micropython/micropython #2678: Unix build: uasyncio __await__ not called

peterhinch, 2016-12-12:
Is this special method supported? The following code behaves differently under CPython and MicroPython.
```python
import uasyncio as asyncio

def genf(n):
    while n:
        print(n)
        yield
        n -= 1

class Foo():
    def __iter__(self):
        for n in range(5):
            print('__iter__ called')
            yield

    def __await__(self):
        print('__await__ called')
        yield from genf(5)

async def bar():
    foo = Foo()
    print('waiting for foo')
    await foo
    print('done')

async def main(delay):
    await asyncio.sleep(delay)
    print("I've seen starships burn off the shoulder of Orion...")
    print("Time to die...")

loop = asyncio.get_event_loop()
loop.create_task(bar())
loop.run_until_complete(main(5))
```
Under CPython the outcome is as expected: ``__iter__()`` is not called, ``__await__()`` is:
```
Python 3.5.2 (default, Dec  2 2016, 15:29:25) 
[GCC 4.8.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rats6
waiting for foo
__await__ called
5
4
3
2
1
done
I've seen starships burn off the shoulder of Orion...
Time to die...
>>> 
```
Under MicroPython the converse is true:
```
MicroPython v1.8.1-925-g46e59c5 on 2016-12-03; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import rats6
waiting for foo
__iter__ called
__iter__ called
__iter__ called
__iter__ called
__iter__ called
done
I've seen starships burn off the shoulder of Orion...
Time to die...
>>> 
```

dpgeorge, 2016-12-13:
No, it's not supported.  The await/async syntax is there but there are still some unimplemented parts in the runtime (like `__await__` support).

peterhinch, 2016-12-13:
OK. Using ``__iter__`` seems to work in that it produces an awaitable class. Using
```python
    def __iter__(self): # workround for issue #2678
        yield from self.__await__()
```
would appear to produce a solution which will continue to work when ``__await__`` is supported. Can you see any hidden pitfalls?

dpgeorge, 2016-12-13:
Per https://www.python.org/dev/peps/pep-0492/, `__await__` is equivalent to `__iter__`.  So I think a way to support CPy and uPy is to implement just `__iter__` and put `__await__=__iter__` in the class body (or the other way around).

peterhinch, 2016-12-13:
OK, thanks!

closed 2016-12-13 by peterhinch