micropython/micropython #2679: Error when using import in main.py on ESP12E

andreybavt, 2016-12-12:
I can't manage to have a startup script (main.py) containing any import.
For example if I save a main.py with the contents of 
```
print(123)
```
then everything works fine, the number is printed at startup, however, if for example I change the code to:
```
import time
print(123)
```
then the error occurs:
```
File "main.py", line 1
SyntaxError: invalid syntax
```

Do you have any clue why that might happen?

closed 2016-12-12 by andreybavt

dpgeorge, 2016-12-13:
Did you intend to close this issue, did you fix the problem?  Possibly you have some non-ASCII chars on the line with the import.  It should work.

tusharvb19, 2017-06-20:
What is the solution???
I am also facing the same error!

dpgeorge, 2017-06-21:
@tusharvb19 if you can provide a full bug report (version of software and hardware used, and minimal code to reproduce the error) then we might be able to work out the issue/solution.