micropython/micropython #2681: stmhal: Option to run from internal oscillator (HSI)

ryannathans, 2016-12-13:
I think it would be a fantastic option if there was an option to compile without an external oscillator and just use the 16MHz internal RC.

Are there any obvious constraints preventing this?

ryannathans, 2016-12-13:
Commenting the below in system_stm32.c and setting MICROPY_HW_CLK_PLLM to 16 in mpconfigboard.h seems to do the trick... 

```
    #if defined(MCU_SERIES_F4) || defined(MCU_SERIES_F7)
    //RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
    //RCC_OscInitStruct.HSEState = RCC_HSE_ON;
    //RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
```

dpgeorge, 2016-12-13:
machine.freq() allows you to select the HSI if the requested frequency is the right value (eg machine.freq(8000000) on pyboard).  So it is supported, just not as a compile-time option.

ryannathans, 2016-12-13:
@dpgeorge for curiosity sake, what dictates whether the HSI will be used or not?

https://github.com/micropython/micropython/blob/ad166857bc93e519bca8c4f14523dcce654a5994/esp8266/modmachine.c#L60

Are these hard coded values a result set/fixed PLL values on the pyboard? If so, I think that should be refactored to support other boards?

For context, I'm working with the f439 board (192MHz)

Anyway, is an acceptable solution for compile-time support of HSI as simple as adding an #if defined USE_HSI and skipping those three lines? (for F4 at least...)

chrismas9, 2016-12-13:
>>Are there any obvious constraints preventing this?

USB requires a 0.25% accurate clock and may be unreliable running from HSI. Only a few newer STM32 parts can PLL sync their RC oscillators to either the USB heartbeat or 32kHz clock.

ryannathans, 2016-12-13:
@chrismas9 fancy, my USB just happens to be working - woo :)

chrismas9, 2016-12-13:
Don't take it to the beach or snow. It's very temperature dependent

dpgeorge, 2016-12-13:
> Are these hard coded values a result of set/fixed PLL values on the pyboard?

Yes.  Note that the HSI is not connected to the PLL, it's just used for a low speed system clock.

> Anyway, is an acceptable solution for compile-time support of HSI as simple as adding an #if defined USE_HSI and skipping those three lines? (for F4 at least...)

I think it needs more than that, especially if you want the PLL to be clocked by the HSI.  Note that this enhancement would be very low priority on my list.

ryannathans, 2016-12-13:
Woah, do you mean my board is actually running at 16MHz now? Will check when I'm home

ryannathans, 2016-12-13:
How can USB be working if HSI isn't going through PLL?

chrismas9, 2016-12-13:
Have you removed the crystal or just switched core clock to HSI? There is a separate 48 MHz PLL for USB. It has its own mux from HSE and HSI. Maybe the 48 MHz PLL is still running from the crystal.

ryannathans, 2016-12-13:
@chrismas9 shorting external oscilator to ground while typing in REPL/accessing files should be a good enough test... Right?

chrismas9, 2016-12-13:
Yeh. HSI will have been factory trimmed for 20 - 25 deg C. If it works for
you that's good but don't expect it to be reliable over all devices over
temperature range or with aging

ryannathans, 2016-12-13:
@chrismas9  I don't want or need USB, there's confusion as to how it's working as all the peripheral clocks are correct yet HSE is grounded. Maybe it is going through PLL?

fmanno, 2019-03-04:
Hi,
I happen to have a very old NUCLEO-F401RE board (revision C-01) and in this version of the board it's not possible to use HSE as a clock source without hardware modifications. A very good summary of the HW differences is [here](https://api.riot-os.org/group__boards__nucleo-f401re.html) and in section 6.7.1 of the [user manual](https://www.st.com/content/ccc/resource/technical/document/user_manual/98/2e/fa/4b/e0/82/43/b7/DM00105823.pdf/files/DM00105823.pdf/jcr:content/translations/en.DM00105823.pdf).

This is not a trivial issue as building and flashing the current NUCLEO_F401RE variant of micropython in one of these C-01 boards result in a non-working system. I agree default should be HSE as the HSI doesn't seem to be very reliable but it's good enough to play around with the board and at least try micropython out. It should be easier to configure HSI as the clock source.

I've created a pull request for this #4571. I've tried it on my C-01 board and I get a REPL (I couldn't get one with the defaults) I don't have any of the other revisions to test so someone should try this on another HW. Also I'm not 100% sure the changes to powerctrl.c. They look sensible to me but they should be reviewed carefully by someone who knows more about the area.

dpgeorge, 2019-03-05:
HSI support was implemented in f938e70c6905474b08f7b5e6df3da9c1b0dcf01a

closed 2019-03-05 by dpgeorge