micropython/micropython #2685: Is micropython for NUCLEO_F767ZI working?

sergiuszm, 2016-12-14:
Hello,

I have seen that you merged support for NUCLEO_F767ZI finally. Unfortunately, I am not able to get it to work.

I am using:
- Linux Mint 18
- GNU Toolchain: arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
- GNU gdb (7.10-1ubuntu3+9) 7.10
- st-util 1.2.0-155-gecc31f7

I am able to compile and flash firmware but that's all. I am not able to access MicroPython REPL via screen or picocom tools. There is only a blinking cursor and nothing more. I can't write anything.

**Code compilation output**
```
make BOARD=NUCLEO_F767ZI 
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-NUCLEO_F767ZI/genhdr
Create build-NUCLEO_F767ZI/genhdr/pins.h
Create stmconst build-NUCLEO_F767ZI/modstm_qstr.h
Create build-NUCLEO_F767ZI/genhdr/pybcdc.inf
Create build-NUCLEO_F767ZI/genhdr/pybcdc_inf.h
Generating build-NUCLEO_F767ZI/genhdr/mpversion.h
GEN build-NUCLEO_F767ZI/genhdr/qstr.i.last
GEN build-NUCLEO_F767ZI/genhdr/qstr.split
GEN build-NUCLEO_F767ZI/genhdr/qstrdefs.collected.h
QSTR updated
GEN build-NUCLEO_F767ZI/genhdr/qstrdefs.generated.h
mkdir -p build-NUCLEO_F767ZI/hal/f7/src/
mkdir -p build-NUCLEO_F767ZI/lib/fatfs/
mkdir -p build-NUCLEO_F767ZI/lib/fatfs/option/
mkdir -p build-NUCLEO_F767ZI/lib/libc/
mkdir -p build-NUCLEO_F767ZI/lib/libm/
mkdir -p build-NUCLEO_F767ZI/lib/mp-readline/
mkdir -p build-NUCLEO_F767ZI/lib/netutils/
mkdir -p build-NUCLEO_F767ZI/lib/timeutils/
mkdir -p build-NUCLEO_F767ZI/lib/utils/
mkdir -p build-NUCLEO_F767ZI/py/
mkdir -p build-NUCLEO_F767ZI/py/../extmod/
mkdir -p build-NUCLEO_F767ZI/py/../lib/embed/
mkdir -p build-NUCLEO_F767ZI/usbdev/class/src/
mkdir -p build-NUCLEO_F767ZI/usbdev/core/src/
CC ../py/mpstate.c
CC ../py/nlrx86.S
CC ../py/nlrx64.S
CC ../py/nlrthumb.c
CC ../py/nlrxtensa.S
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/reader.c
CC ../py/lexer.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmbase.c
CC ../py/asmx64.c
CC ../py/emitnative.c
CC ../py/asmx86.c
CC ../py/emitnative.c
CC ../py/asmthumb.c
CC ../py/emitnative.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
[... 310 lines cut ...]

port is        : /dev/ttyACM0
flowcontrol    : none
baudrate is    : 9600
parity is      : none
databits are   : 8
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Terminal ready
```

So, i get `Terminal ready` and that's all - I can't do anything else.

Is there anyone who get it to work with this board?

Thanks for any information.

dpgeorge, 2016-12-14:
I was able to get a REPL for this board.  There are 2 ways:
- via st-link's VCP pass through on the st-link USB connector; this is wired to UART3 on the STM32F767, and the REPL appears on this UART; you'll need to explicitly set the baudrate to 115200 when using this port
- via the other USB connector, the one connected directly to the STM32F767, in USB FS mode; this is the recommended way to access the REPL

sergiuszm, 2016-12-15:
Thanks @dpgeorge you are right. VCP works perfectly :)

closed 2016-12-15 by sergiuszm