micropython/micropython #2691: lwip setsockopt() unimplemented

nickovs, 2016-12-16:
The current implementation of `lwip_socket_setsockopt()` basically only handles `SOF_REUSEADDR` and doesn't support any of the other options that the underlying lwip library supports. As a result things like multicast (which would be useful for device discovery), TTL, keep-alive and various other functions that would be useful.

This issue is to record the fact that this work needs doing and discuss any design ideas for the implementation.

nickovs, 2016-12-16:
Note that in the comments on #2198 @mjkillough suggested that he might look into implementing this but it looks like the problem in hand got resolved another way. If someone else is working on this then please comment. There is a small chance that I'll find the time, but only small!

nickovs, 2016-12-17:
Having taken a closer look at [modlwip.c](https://github.com/micropython/micropython/blob/master/extmod/modlwip.c), putting in support for socket options is far from simple because of the very odd way that this was written. The full getsockopt/setsockopt code can only use used if a socket has a socket number, and these sockets don't.

The current implementation bypasses the higher level lwip API from lib/lwip/src/include/lwip/sockets.h and instead piles into the morass of the internals of lwip, reimplementing a bunch of the otherwise opaque internals in a way that prevents use of most of the higher level functions.

Can I ask the authors ( @dpgeorge @pfalcon @galenhz ) why this code doesn't use the top level socket API? Most of the memory overhead of the socket table is statically allocated so you're already paying for it. It seems like it's just adding a stack of complexity and tying the code to a particular internal implementation of lwip in return for hardly any memory saving and the removal of a table lookup and a pointer de-reference.

galenhz, 2016-12-17:
The high-level socket API of lwip adds other baggage as well; it basically (from what I remember) requires a number of RTOS bits (like messaging and threads) in order to work. I was working on the STM version, which has no RTOS. That's why I had to end up using the lowest-level API and implementing a kinda-sorta implementation of sockets on top of it.

markperdue, 2017-01-31:
@nickovs have you had a chance to work on any updates to this? I am also running into an issue trying to set up multicast on an esp8266. As you know, It seems that the current implementation of setsockopt only accepts ints for the value param.

It would be great to have support for multicast.

nickovs, 2017-02-01:
Unfortunately I've not had a chance to dig into this. It seems that implementing a full setsockopt() is going to be rather a lot of work due to the fact that the existing code makes use of the lightweight, low-level API rather than using the posix-like socket API. I suspect that since the function of setsockopt() that most people need is the handling of multicast, the best option is to reimplement just the bits of setsockopt() necessary for that function and not try to use the higher level APIs.

Having said that, is your need for full multicast, or do you actually need mDNS? The lwip library has a complete mDNS library which can run in the background without needing you handle packets in the Python code. Irrespective of if we implement enough of setsockopt() to handle multicast I think putting a wrapper around the lwip mDNS library would be a "good thing".

Vincent-Ma, 2017-04-07:
Hello,
For a project I am currently working on, I added some functionalities in the [setsockopt](https://github.com/micropython/micropython/blob/master/extmod/modlwip.c#L1066) function to support multicast (inspired from this 3 library files in the esp8266 Arduino repository:  [WiFiUdp.cpp](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiUdp.cpp#L103), [ESP8266SSDP.cpp](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266SSDP/ESP8266SSDP.cpp#L179), [ESP8266mDNS.cpp](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/ESP8266mDNS.cpp#L196)). In my case it is for a device discovery system, and the socket is only used for receiving data. 

The modified modlwip.c is available [here](https://gist.github.com/Vincent-Ma/7943007d346ad030888d37e0e6ae55fc) ([diff](https://www.diffchecker.com/oEmLLxVp))

Also to get this working I added some constant (`socket.IPPROTO_IP`, `socket.IP_ADD_MEMBERSHIP`, and `socket.INADDR_ANY`), and the static method `socket.inet_aton()`.

You can test the modification with the following code:

```python
import socket
import ustruct

MCAST_GRP = '239.255.255.250'
MCAST_PORT = 1900

mreq =  ustruct.pack("4s",  socket.inet_aton(MCAST_GRP)) + ustruct.pack("l", socket.INADDR_ANY)
sockReceiver = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

sockReceiver.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
sockReceiver.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, mreq)
sockReceiver.bind((MCAST_GRP, MCAST_PORT))

# Receive ssdp requests
sockReceiver.recvfrom(1024)
```

I am using a Nodemcu Esp8266 (MicroPython v1.8.7-586-gfaf333c-dirty on 2017-04-06).

markperdue, 2017-04-15:
@Vincent-Ma a big thanks to you for your changes. I was able to join a multicast group and receive messages with them.

nickovs, 2017-04-15:
Thanks Vincent! Have you submitted a pull request to get this into the main-line code?

Cheers,
Nicko

pfalcon, 2017-04-15:
It's impossible to get that code into the mainline - it's derived from the code with incompatible license. Whoever would want to get some code into mainline, will need to make a clean-room implementation without deriving it from incompatible sources and sign off on the clean-room methodology (and to ensure it, I'd personally recommend to avoid even looking at @Vincent-Ma's branch).

nickovs, 2017-04-15:
On 15 Apr 2017, at 15:19, Paul Sokolovsky <notifications@github.com> wrote:

> It's impossible to get that code into the mainline - it's derived from the code with incompatible license. Whoever would want to get some code into mainline, will need to make a clean-room [...]

Aha. OK. It's a good thing I hadn't taken a look at this yet!

Cheers,
Nicko

Vincent-Ma, 2017-04-16:
Hum, ok sorry I haven’t considered the license issue. I will be more careful in the future.
However, the inspired part concern these lines:
````c
igmp_joingroup(&ifaddr, &multicast_addr)

socket->pcb.udp->multicast_ip = multicast_addr;
````
Which are only LWIP function and structure. So, someone with only knowledges in LWIP could have done the same implementation. Does that count as derived license?

pfalcon, 2017-04-17:
@Vincent-Ma this matter (licensing due diligence) was recently discussed at https://github.com/micropython/micropython-esp32/issues/52#issuecomment-289439586 .

My reply above is the formal reply in the same vein as in the thread above: you stated that you based your code on the "esp8266 Arduino repository" which is known to have LGPL license, a maintainer clearly stated that such code is not suitable for the mainline.

This reply is addressed not just to you, but to any party which will study the licensing due diligence of MicroPython project (which would a prerequisite for any serious usage of the project by a commercial entity). As it's an OpenSource project, this study will happen by looking for precedents in the public communication of the project, like this ticket. And as clear from this and the other thread, the project maintainers adopt highly conservative policy in this regard, that any code suspicious of licensing issues is not suitable for the mainline.

> Does that count as derived license?

I can share my opinion, but in general, only a licensing lawyer could provide any warranty behind an opinion, and opinion of one licensing lawyer can be challenged by another licensing lawyer in a licensing court. Unless you want to include indemnification for each and every MicroPython user together with your code contribution, please kindly adopt the same conservative approach as the project maintainers and don't base your code on anything with incompatible license.

In this case, the best solution would be for someone else to develop a code from scratch, based on the information that "only LWIP function and structure" are required. That should not require much effort, as the functionality is fairly trivial, and 90% of the effort would go towards testing that it works as expected, not to (re)writing code.

Thanks.

Timmenem, 2017-05-23:
Hey Guys,
I'm currently trying to get multicast to work, too. I would really like to use some code from
[lwip/api/sockets.c](https://github.com/pfalcon/esp-open-lwip/blob/8c39d2179a273553466043f388772abb6251a4ca/lwip/api/sockets.c#L2148) in order to get the `IP_ADD_MEMBERSHIP` case to work and be able to `igmp_joingroup()`.

Unfortunately, I can't successfully extract the IP information that is passed to the current `lwip_setsockopt` function as `args[3]` in [modlwip.c](https://github.com/micropython/micropython/blob/master/extmod/modlwip.c#L1066).

My current code that deals with the `IP_ADD_MEMBERSHIP` case:
```
struct ip_mreq *imr = (struct ip_mreq *)(args[3]);

ip_addr_t if_addr;
ip_addr_t multi_addr;

inet_addr_to_ipaddr(&if_addr, &imr->imr_interface);
inet_addr_to_ipaddr(&multi_addr, &imr->imr_multiaddr);

igmp_joingroup(&if_addr, &multi_addr);
```

But whatever I try, i can't get `if_addr` and `multi_addr` to contain the correct IPs that I pass to my python scripts. I found a couple of examples online on how to struct.pack() the two IP adresses in order to pass them as a single parameter when calling `setsockopt()`, but no matter which way I try, `igmp_joingroup() `always returns an error and I am very sure that this happens because of "bad" IP adresses in the `ip_mreq` struct.

I already tried the following approaches in order to get a proper "mreq" to pass in python:
```
MCAST_GRP = '239.255.255.250'   
mreq =  ustruct.pack("4s",  socket.inet_aton(MCAST_GRP)) + ustruct.pack("l", socket.INADDR_ANY)
```
```
group = socket.inet_aton(MCAST_GRP)
mreq = ustruct.pack('4sL', group, socket.INADDR_ANY)
```

```
addrinfo = socket.getaddrinfo(group, MYPORT)[0]
group_bin = socket.inet_aton(addrinfo[4][0])
mreq = group_bin + ustruct.pack('l', socket.INADDR_ANY)
```

Right now I believe that there might be something wrong with my approach to apply `struct ip_mreq` to 
`args[3]` as I am not really sure about its data type (`mp_obj_t`) and what I can actually do with it.

Any help would be greatly appreciated :o)

Timmenem, 2017-05-23:
I got some success with my approach. If i use `mp_obj_str_get_data()` on the `args[3]` that is provided as `optval`, I can assign the result to IGMP `ip_mreq` struct. As a result, the struct then holds the 2 IP adresses.

I can then pass that struct to the code I took from lwip's sockets.c
```
size_t len;
const char *ips = mp_obj_str_get_data(args[3], &len);
struct ip_mreq *imr = (struct ip_mreq *)ips;

ip_addr_t if_addr;
ip_addr_t multi_addr;

inet_addr_to_ipaddr(&if_addr, &imr->imr_interface);
inet_addr_to_ipaddr(&multi_addr, &imr->imr_multiaddr);

if (igmp_joingroup(&if_addr, &multi_addr) != 0) {
    mp_raise_OSError(MP_EIO);
}

inet_addr_to_ipaddr(&socket->pcb.udp->multicast_ip, &imr->imr_multiaddr);
```

Now I can subscribe to a multicast group and receive messages with this python code:

```
import socket
import ustruct

try:
    from socket import sockaddr
except:
    sockaddr = lambda x: x

MCAST_GRP = '238.255.255.249'
MCAST_PORT = 1900

mreq =  ustruct.pack("4s",  socket.inet_aton(MCAST_GRP)) + ustruct.pack("l", socket.INADDR_ANY)

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
sock.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, mreq)
sock.bind((MCAST_GRP, MCAST_PORT))

while True:
    data, addr = sock.recvfrom(1024)

    print("From: ", sockaddr(addr))
    print(data)
```

jonnor, 2024-07-27:
Is this issue still relevant, or can it be closed? There seems to have been at least two merge requests for functionality in this area merged.

robert-hh, 2024-07-27:
Outdated. Please re-open if needed.

closed 2024-07-27 by robert-hh