micropython/micropython #2692: RFC modframebuf.c scrolling behaviour

peterhinch, 2016-12-17:
Currently ``framebuf_scroll`` does not clear the region of the display which is exposed by scrolling. While I appreciate that this behaviour is by design, I'd submit that it's seldom what is actually required.

An option would be good, perhaps via a colour arg:
 1. Behave as at present leaving exposed region unchanged.
 2. Fill with a colour.

Any thoughts, or reasons why the current behaviour has been chosen?

mkarliner, 2016-12-17:
+1 for just clearing, unless anyone has a large volume of legacy 
code that depends upon the current behaviour.

peterhinch, 2016-12-17:
A colour arg with a 'no colour' default (say -1) could be used to maintain compatibility.

hoihu, 2016-12-19:
maybe framebuf needs a background value at instantation time (which can be used by several methods, not just scroll - e.g text)? 

I favour a change towards clearing. The framebuf module is still relatively new... I'd vote for a backward incompatible change since that should be more space efficient.

dpgeorge, 2016-12-21:
> reasons why the current behaviour has been chosen?

Simply because it was the simplest thing to do, and is the most general (you can clear if you need, but you can't un-clear).

> An option would be good, perhaps via a colour arg:
> 
>     Behave as at present leaving exposed region unchanged.
>     Fill with a colour.

Adding a "fill" arg would be ok, for the colour to fill with.  `None` could be used to specify no fill, and could be the default.