micropython/micropython #2693: stmhal: machine.Timer not available in stmhal

BrendanSimon, 2016-12-17:
According to this [page](http://docs.micropython.org/en/latest/wipy/library/machine.Timer.html) the `machine` module should have a `Timer` class.

However, it doesn't appear to exist on the `stmhal`.

```
PYB: soft reboot
MicroPython v1.8.6-204-g2daa866 on 2016-12-17; OLIMEX STM32-E407 with STM32F407
Type "help()" for more information.
>>> import machine
>>> t = machine.Timer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'Timer'
>>> dir(machine)
['__name__', 'info', 'unique_id', 'reset', 'soft_reset', 'bootloader', 'freq', 'rng', 'idle', 'sleep', 'deepsleep', 'reset_cause', 'disable_irq', 'enable_irq', 'time_pulse_us', 'mem8', 'mem16', 'mem32', 'Pin', 'I2C', 'SPI', 'WDT', 'PWRON_RESET', 'HARD_RESET', 'WDT_RESET', 'DEEPSLEEP_RESET', 'SOFT_RESET']
```

mcauser, 2017-01-17:
I believe the stmhal port is currently using `pyb.Timer`
http://docs.micropython.org/en/latest/pyboard/pyboard/tutorial/timer.html

dpgeorge, 2019-10-31:
`machine.Timer` was added to stm32 in 48b25a841b04108d4dc399fad87b413f9f52c6bb

closed 2019-10-31 by dpgeorge