micropython/micropython #2694: Variable not found when using micropython.native code emitter

smurfendrek123, 2016-12-17:
So my main.py file imports the file Encryption, that file imports Subclasses_Encryption and i added the @micropython.native in some functions in subclasses. But now it's saying that a certain variable (its a matrix called S block) is not defined, while it worked without the @micropython.native.

I did it like this: https://github.com/smurfendrek123/slimme-thuiszorgmonitor/blob/master/hulpcode/nodigefiles/Subclasses_Encryptienative.py

I've been told to file a bug report, but it might also be my fault, i'm not very experienced with this.

robert-hh, 2016-12-17:
@micropython.native code functions seem not to see global symbols. I stumbled over that too. I do not know whether it's a bug. In my case, I just moved that variable into the function using it. That may be possible here too. Otherwise, you could try to supply s_blok and inv_s_block as function parameters,

smurfendrek123, 2016-12-17:
Thanks, i'll try that!

robert-hh, 2016-12-18:
That could caused by the same reason: no visibility of global symbols. You could try to put that into a class and access the other objects/functions through self.xxx. I just tried that, and it seems to work. The sample code below can access the data & function pointer defined in init
```
class tgl:

    def __init__(self):
        import machine
        pd_sck=4
        self.pdsckPin = machine.Pin(pd_sck, machine.Pin.OUT, value=0)
        self.enable = machine.enable_irq
        self.disable = machine.disable_irq

    @micropython.native
    def toggle(self):
        my = 0
        for i in range(24):
            isr = self.disable()
            self.pdsckPin.value(1)
            self.pdsckPin.value(0)
            self.enable(isr)

tc = tgl()
tc.toggle()

```

dpgeorge, 2016-12-19:
Thanks for the report.  This is a bug and is actually a duplicate of #1573.

The workaround is as @robert-hh suggested: don't use globals with native code, store them elsewhere.

closed 2016-12-19 by dpgeorge