micropython/micropython #2695: ESP8266 Assembler: Offset calculation wrong in branches?

robert-hh, 2016-12-18:
MicroPython v1.8.6-202-g56a514c-dirty on 2016-12-18; ESP module with ESP8266

I have a problem with this little piece of test code:
```
def loop():
    import machine
    machine.Pin(4, machine.Pin.OUT, value=0)
    do_loop()
    print("loop done")

@micropython.asm_xtensa
def do_loop():
    movi(a2, 0x60000300)
    movi(a5, 0x10)
    movi(a3, 10) # loop count
    movi(a4, 1)
    label(loop_start)
    s8i(a5, a2, 4)
    s8i(a5, a2, 8)
    sub(a3, a3, a4)
    bnez(a3, loop_start)
```
If I take it as it is, the esp simply crashes. If I change line 124 of py/asmxtensa.c to
`    int32_t rel = dest - as->base.code_offset - 0;`
which matches the output of the xtensa assembler, the loop is executed as expected, but the esp crashes after that. Did I change registers which I should not?

dpgeorge, 2016-12-19:
The reason is that your code generates an l32r instruction which requires a constant table, and the size of the constant table (here being 4 bytes) is not being computed correctly, hence the labels are off by 4 bytes.  This should be fixed by de9cd00b39fbd66279dda69bc642ac2f3c459fa1.

closed 2016-12-19 by dpgeorge

robert-hh, 2016-12-19:
Hello Damien. Thanks a lot. The assembler version works. Now, the equivalent viper version runs into reset, which used to work before:
```
def loop():
    import machine
    machine.Pin(4, machine.Pin.OUT, value=0)
    do_loop()
    print("Done")

@micropython.viper
def do_loop():

    GPIO_OUT_SET = ptr32(0x60000304) # GPIO Output set
    GPIO_OUT_CLR = ptr32(0x60000308) # GPIO Output clear
    for i in range(10):
        GPIO_OUT_SET[0] = 0x10 # set bit 4
        GPIO_OUT_CLR[0] = 0x10 # clear bit 4
```
MicroPython v1.8.6-205-g91359c8-dirty on 2016-12-19; ESP module with ESP8266
Error Report:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 32012, room 16 
tail 12
chksum 0xc4
ho 0 tail 12 room 4
load 0x3ffe8000, len 1080, room 12 
tail 12
chksum 0x33
ho 0 tail 12 room 4
load 0x3ffe8440, len 3000, room 12 
tail 12
chksum 0xf9
csum 0xf9

```

dpgeorge, 2016-12-19:
See #2705.