micropython/micropython #2697: ESP8266 wifi station scan hidden SSID

Adam5Wu, 2016-12-19:
I have trouble getting network with hidden SSID scanned, even with latest nightly build.
After reading SDK documents, it seems that the problem is at the call to wifi_station_sacn.
```
wifi_station_scan(NULL, (scan_done_cb_t)esp_scan_cb);
```
On older document, it says pass null as config in the first parameter to list all APs, but that version of SDK didn't have the 'show_hidden' option in its config structure;
More modern version of SDK added this 'show_hidden' option, but my guess is they left default value as 'false' (resonable choice), but didn't update the document.

As a result, if config is null, no hidden SSID will ever be enumerated, although the python part of the API fully capable of handling hidden SSID...

The fix would be, to always pass in a proper config structre. If user can pass in a parameter to choose whether they wany hidden SSID, it would be even better!

I didn't have enough free time to setup dev environment, so if someone can quickly test the fix it would be great. Otherwise, I may find some time in the coming holiday weeks and maybe generate a PR later :)

Adam5Wu, 2016-12-26:
Setup the dev environment tonight, my hypothesis and proposed fix is now verified.

Fix is easy, just three lines. Submitted PR #2723 

Happy holidays! :D

dpgeorge, 2021-08-03:
See #7611 for scanning for hidden SSIDs.

dpgeorge, 2021-08-07:
Scanning hidden APs was enabled in 8616129f2e382310802950db066178648a9429a3

closed 2021-08-07 by dpgeorge