micropython/micropython #2698: Hi I am using Micro python to OTA updates on cc3200 device i am getting error at telnet login

theja0473, 2016-12-19:
[image]
[image]

What might be the error shows did any one got this kind of error.

dpgeorge, 2016-12-19:
Using `make deploy` in the cc3200 directory requires Python 3.4 or greater.  If you have python3 installed then change line 1 of cc3200/tools/update-wipy.py to have `python3` at the end of the line instead of `python`.

pfalcon, 2017-04-05:
Hopefully the above cleared the matter. If not, please reopen, providing detailed info about your environment.

closed 2017-04-05 by pfalcon