micropython/micropython #2700: ESP8266 Build failure

peterhinch, 2016-12-19:
I updated and recompiled the toolchain and attempted to build the current source with the following outcome:
```
[lines omitted]
CC ../drivers/dht/dht.c
CC build/frozen_mpy.c
LINK build/firmware.elf
xtensa-lx106-elf-ld: build/firmware.elf section `.irom0.text' will not fit in region `irom0_0_seg'
xtensa-lx106-elf-ld: region `irom0_0_seg' overflowed by 8088 bytes
make: *** [build/firmware.elf] Error 1
Build failure
/mnt/qnap2/data/Projects/MicroPython/micropython/esp8266
[adminpete@axolotl]: /mnt/qnap2/data/Projects/MicroPython/micropython/esp8266
$
```

dpgeorge, 2016-12-19:
Do you have any extra frozen code in scripts/ or modules/?  If you increase the `len` argument at line 8 of esp8266.ld to something like 0x90000 it should build (but won't run because the firmware will clash with the filesystem), and then you can report the final size of the binary to give more info about the problem.

robert-hh, 2016-12-19:
Hello Pete. I ran into the same problem - too much code to be frozen. As a temporary workaround I removed Berkeley-DB from the build, by setting MICROPY_PY_BTREE = 0 in Makefile.

peterhinch, 2016-12-20:
Ah! I should have thought of that. Thanks both.

closed 2016-12-20 by peterhinch

chrisbalmer, 2016-12-24:
Is it possible to expand the firmware size at the expense of the file system? I'm hitting the same wall just trying to add http.client and urllib.parse to the esp8266 as frozen byte code in the firmware. The module I'm using has 4MB (32Mbit) flash.

robert-hh, 2016-12-25:
You can do so, like Paul showed as an example:
> I see @pfalcon changed the firmware image size to 0x90000 at https://github.com/micropython/micropython/commit/3372f6958653cd920d07de24f1e0284727a14237, and there was a subsequent change to set the start of the file system at 3c9510d.

The first commit increases the ROM length by 7*0x1000 bytes, the second decreases the file system size by 7 sectors of 0x1000=4096 bytes length each. You may change the numbers to match your needs. You have to rebuild the file system, and you should erase the flash before you start.
I was too lazy to do that, when I ran into that memory gap, and simply removed the Berkeley-DB from the build.

peterhinch, 2016-12-26:
The ideal might be a compile time option whereby users of frozen bytecode could specify the space to reserve for firmware. You'd need to erase flash if you changed it.

dpgeorge, 2016-12-29:
There is a need to support modules with 512k, 1M and >1M of flash.  The default is 1M, hence the reason for restricted firmware size.  It would make sense to have an ld script for >1M modules with more room for the firmware.

peterhinch, 2017-03-20:
To anyone with this problem flashbdev.py has changed since the above. It now determines the filesystem start from the esp module. I did the following to make plenty of space on the 4MB reference board:

esp8266/esp8266.ld line 8 now reads
```
    irom0_0_seg :  org = 0x40209000, len = 0xa7000
```
and esp8266/modesp.c now has
```C
STATIC mp_obj_t esp_flash_user_start(void) {
    if (IS_OTA_FIRMWARE()) {
        return MP_OBJ_NEW_SMALL_INT(0x3c000 + 0x90000);
    } else {
        return MP_OBJ_NEW_SMALL_INT(0xb0000);
    }
}
```