micropython/micropython #2701: ESP8266 RTC.irq method rejects arguments. Missing RTC constants in machine module.

peterhinch, 2016-12-19:
Testing was done with a build using code of 27th Nov. I couldn't update because of https://github.com/micropython/micropython/issues/2700. Pasting the following:
```python
import machine
import micropython
micropython.alloc_emergency_exception_buf(100)

def alarm_handler(rtc_o):
   print("Hello from alarm_handler")

rtc = machine.RTC()
rtc.alarm(machine.RTC.ALARM0, 10000)
rtc_i = rtc.irq(trigger=machine.RTC.ALARM0, handler=alarm_handler, wake=machine.DEEPSLEEP)
```
produced (from rtc_i = rtc.irq(args))
```
Traceback (most recent call last):
  File "<stdin>", line 10, in <module>
TypeError: extra keyword arguments given
>>> 
```
The method fails with a variety of combinations of default and explicit args.

The constants IDLE and SLEEP are missing from the machine module.

skbobade, 2017-04-08:
Any update on this? I still am getting same error with the latest code.

dpgeorge, 2017-04-11:
> Further testing indicates that it objects to the handler arg. Is this not yet implemented?

That's correct, the "handler" arge is not yet implemented.  Same with waking from IDLE and SLEEP (so those constants are not provided).