micropython/micropython #2704: Driver for BME280 broken since v1.8.6-77-ga3320e7

jurviz, 2016-12-19:
I had to revert back to above level on ESP8266 to get BME280 driver working with I2C.
I had applied the change to I2C (-1 as 1st parameter), but could not get read_from_mem to work.
The function complained about missing positional parameter.
Also the doc for I2C on ESP8266 is aout of sync.

dpgeorge, 2016-12-20:
With the refactors to I2C that were done recently, there were very few changes that impacted existing code, the changes were mostly backwards compatible.

> I had applied the change to I2C (-1 as 1st parameter)

You don't need to do this, it still works as it did before, ie: `i2c = machine.I2C(scl=Pin(5), sda=Pin(4))`

> but could not get read_from_mem to work.

Please elaborate on this.  I just tried it and it works as per the docs, eg: `i2c.readfrom_mem(0x6b, 0xa8, 6)` (that's for an L3GD20H).

> Also the doc for I2C on ESP8266 is aout of sync.

Please eloborate on this.

jurviz, 2016-12-20:
I will give all the details when I'm back home after Christmas.

jurviz, 2016-12-29:
The code i used before:
from machine import Pin, I2C
scl = Pin(5)
sda = Pin(4)
i2c = I2C(scl, sda)
gives error message
ValueError: invalid I2C peripheral
It works using keyword arguments.
i2c = I2C(-1, scl, sda) 
or
i2c = I2C(scl=scl, sda=sda)
The docs don't give a hint of a positional argument.

jurviz, 2016-12-29:
The readfrom_mem works as documented. The error stems from int.from_bytes complaining:
TypeError: function missing 1 required positional arguments.
So this is unrelated to I2C.

jurviz, 2016-12-29:
Found that int.from_bytes now requires 'little' as 2nd argument.

dpgeorge, 2016-12-30:
> It works using keyword arguments.
> i2c = I2C(-1, scl, sda)
> or
> i2c = I2C(scl=scl, sda=sda)
> The docs don't give a hint of a positional argument.

Yes, you are right, some of the docs were not updated to reflect the changes, sorry about that!  I have fixed the docs in d377c837946dc7476918dcda1580467396e5a153.

> Found that int.from_bytes now requires 'little' as 2nd argument.

Yes, that was a bug that was fixed recently, so that int.from_bytes and int.to_bytes are compatible with how CPython works.

closed 2016-12-30 by dpgeorge