micropython/micropython #2705: esp8266 viper: simple test program is instable

robert-hh, 2016-12-19:
Hello Damien. Thanks a lot for fixing the assembler version. Now, the equivalent viper version runs into reset, which used to work before:
```
def loop():
    import machine
    machine.Pin(4, machine.Pin.OUT, value=0)
    do_loop()
    print("Done")

@micropython.viper
def do_loop():
    GPIO_OUT_SET = ptr32(0x60000304) # GPIO Output set
    GPIO_OUT_CLR = ptr32(0x60000308) # GPIO Output clear
    for i in range(10):
        GPIO_OUT_SET[0] = 0x10 # set bit 4
        GPIO_OUT_CLR[0] = 0x10 # clear bit 4
```
Sometimes, it can be called a few times before it breaks.
MicroPython v1.8.6-205-g91359c8-dirty on 2016-12-19; ESP module with ESP8266
Error Report:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 32012, room 16 
tail 12
chksum 0xc4
ho 0 tail 12 room 4
load 0x3ffe8000, len 1080, room 12 
tail 12
chksum 0x33
ho 0 tail 12 room 4
load 0x3ffe8440, len 3000, room 12 
tail 12
chksum 0xf9
csum 0xf9
```
A simpler viper code which just adds up a number in a loop and returns the value is stable.
P.S.: I have written that as a comment in the closed previous issue, but I do not know whether this is noticed.

dpgeorge, 2016-12-20:
I can't reproduce the error but it could be due to garbage collection collecting objects that are referenced by the native code.  This GC tracing problem is now fixed by 0d32f1aeb3e3792f2bd04c499a6d5a22c2efdf0c.  @robert-hh does it fix the problem for you?

robert-hh, 2016-12-20:
Thanks. This morning's build seems to have fixed the problem. The code looks stable now. GC as the cause seems plausible, since it sometimes worked for a while, sometimes failed immediately.

closed 2016-12-20 by robert-hh