micropython/micropython #2706: How to get the infomation of wifi devices, such as mac

xiaocaiyidie, 2016-12-20:
I  use `wlan.scan()`  and get `bssid `.  
On the document, it says:`bssid is hardware address of an access point, in binary form, returned as bytes object. You can use ubinascii.hexlify() to convert it to ASCII form.`
So,i code my program like this.
But,   i got  `b'00873632e71a'` instead of the real mac string 'EC:55:F9:70:D8:45'

Could anyone help me? Thanks!

dpgeorge, 2016-12-20:
You can use `wlan.config('mac')` to get the MAC address of your local device.  The bssid from `scan()` will be the address of the access point.

xiaocaiyidie, 2016-12-20:
Thanks a lot!
 By the way,I used `ap = network.WLAN(network.AP_IF)` .and set config by 
```python
config(essid='micropython',password='micropython')
```
then i connected my phone to this access point. how can i get the `mac` of the phone?
PS(micropython is running on esp8266)

dpgeorge, 2016-12-20:
> how can i get the mac of the phone?

There is no way to do this in Python, and I don't even know if the ESP8266 SDK provides a way to list attached devices (?).

Anyway, I assume you can now get the MAC of your ESP8266 and this issue can be closed.  In the future please ask such questions at http://forum.micropython.org and use github here for bug reports.  Thanks!

closed 2016-12-20 by dpgeorge