micropython/micropython #2707: Proposal: Troubleshooting Wiki

hiway, 2016-12-20:
I recently stumbled across this issue: https://github.com/micropython/micropython/issues/2700 - which I had myself faced a while ago and realized that we could save some time and efforts for the users as well as the core developers by having common gotchas compiled in one place where it is easy to look up the situation you are in by eyeballing through the headers.

Here is an attempt/poc: https://github.com/micropython/micropython/wiki/Troubleshooting

Thoughts? 

*If I missed another place where such information is already documented, perhaps it needs more visibility*

dpgeorge, 2016-12-21:
It's a good idea, but the general problem is that people don't like to read documentation :)  And even if they do, a document that lists all problem would be very long and then even less likely to be read, or you may just miss the thing your looking for (eg it appears under some other name/title).

This general problem is solved by things like stackoverflow by having 1) lots of very specific questions and answers; 2) content provided by the community; 3) a good search engine indexing the content.

hiway, 2016-12-21:
I was thinking in limited scope of this repo when looking for a solution - StackOverflow definitely is a better choice. Also, thank you for the reminder that I should use it more :)

Cleaning up the new wiki page and closing this issue. Thank you.

closed 2016-12-21 by hiway